mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #118: Publish PV data to EVCC via MQTT

Hoellenwesen, 2023-06-23:
Hi,

for all of you which are using this great "integration" for Modbus TCP into HA and also would like to use the EVCC addon. 

I wrote some MQTT and EVCC configurations for that you do not have to use a Modbus Proxy to provide the required data. For those who are interested, you can find it here:

https://github.com/Hoellenwesen/home-assistant-configurations/tree/main/evcc

btota, 2023-07-01:
Hey, glad to hear that,  my constellation: Sungrow -->SH10RT+SBR128 Battery+AC011E-01 Wallbox

everything is working fine with mkaiser (thanksalot) and modbus-proxy addon on hassos; except the Sungrow AC011E-01 wb. I don´t have a clue, which sensors to use, so I installed on hassos ModbusTCP2MQTT, Mosquitto broker to get more registers recognized. With no success.On isolarcloud the AC011E-01 is shown up the right way. 

In evcc addon unfortunately I'm not able to use the modbus-proxy (localhost:502), so I tried your automation.yaml and evcc.yaml  `Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2023/07/01 18:29:38 evcc 0.118.2
[main  ] INFO 2023/07/01 18:29:38 using config file: /config/evcc.yaml
[main  ] INFO 2023/07/01 18:29:38 starting ui and api at :7070
[mqtt  ] INFO 2023/07/01 18:29:38 connecting evcc-1293083007 at tcp://localhost:1883
[main  ] FATAL 2023/07/01 18:29:38 failed configuring mqtt: error connecting: network Error : dial tcp [::1]:1883: connect: connection refused
[main  ] FATAL 2023/07/01 18:29:38 will attempt restart in: 5m0s` with issues?

mkaiser, 2023-07-03:
@Hoellenwesen: awesome, took a quick peak at the code. This is exactly, what is missing for a smooth Sungrow --> HA --> EVCC integration. 

I am short on time at the moment, but will try that in a few week and then link your project in my readme :)

@ [btota](https://github.com/btota) maybe post this in https://github.com/Hoellenwesen/home-assistant-configurations/tree/main/evcc ? So it will be more structured hare

btota, 2023-07-03:
> @Hoellenwesen: awesome, took a quick peak at the code. This is exactly, what is missing for a smooth Sungrow --> HA --> EVCC integration.
>
> I am short on time at the moment, but will try that in a [...]

you're right, done :)

closed 2024-12-29 by github-actions[bot]