mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #119: Index out of range issue

lampi84, 2023-07-01:
Since some weeks i'm getting the following error. It was working fine before and i didn't change anything.
Does anyone have an idea what is causing this issue?

Here are some logs:
2023-07-01 22:10:25.669 ERROR (SyncWorker_2) [pymodbus.factory] index out of range
2023-07-01 22:10:25.671 ERROR (SyncWorker_2) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-07-01 22:10:25.671 ERROR (SyncWorker_2) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-07-01 22:10:25.681 ERROR (SyncWorker_0) [pymodbus.factory] index out of range
2023-07-01 22:10:25.681 ERROR (SyncWorker_0) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request

mkaiser, 2023-07-03:
Hi, 

can you give us more information? e.G., which inverter you are using and which versions (inverter, Home assistant, etc.)

are you using a modbus proxy or directly accessing via TCP ?

lampi84, 2023-07-03:
Hi,

I`m using a Sungrow sh6.0rt Inverter. I'm connecting via TCP using your ModBus Home Assistant Integration. It was working fine, but all of a sudden it stopped working.

mkaiser, 2023-07-03:
hmm no clue at the moment - 
standard support question: did you make a clean restart with some minutes power down of the inverter?

lampi84, 2023-08-04:
I restarted everything, but i'm still getting the following error:

2023-08-04 22:11:08.081 ERROR (SyncWorker_0) [pymodbus.factory] index out of range
2023-08-04 22:11:08.083 ERROR (SyncWorker_0) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2023-08-04 22:11:08.083 ERROR (SyncWorker_0) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
2023-08-04 22:11:08.092 ERROR (SyncWorker_4) [pymodbus.factory] index out of range
2023-08-04 22:11:08.092 ERROR (SyncWorker_4) [pymodbus.transaction] Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 208, in execute
    self.client.framer.processIncomingPacket(response,
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 153, in processIncomingPacket
    self._process(callback)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")

Is there any possibility to adjust the code so that i can see why it is failing. Somehow a way to get more advanced logging.

mkaiser, 2023-08-07:
here is something about logging. Can you try it?
https://community.home-assistant.io/t/modbus-detailed-error-information/584790

My other approach would be to comment a bunch of sensors and do try/error

LapplandsCohan, 2023-10-18:
The error looks very similar to what I got when I upgraded Home Assistant Core from 2023.9.2 to 2023.10.2. Didn't have time for troubleshooting so I restored the 2023.9.2 backup and everything worked again.
I will try to upgrade next time I can take the time for troubleshooting in case I get the same problem again.

closed 2023-10-18 by mkaiser