mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #128: Scan interval changes on HA 2023.7

ajbatchelor, 2023-07-15:
I updated to 2023.7 and the scan interval on the entities created by this (amazing) integration change from their default to only updating every 20 minutes.  

For example in the attached screen, the 'load power' entity has scan interval on 10 seconds set in config.yaml, this has been working as it should. Then when I update to 2023.7 it extends out to only updating every 20 mins.

I rolled back to 2023.6 and everything works again.  I can't find anyone else reporting this issue.

[image]

mkaiser, 2023-07-15:
hi, 

this is really strange. I just updated from 23.7.1 to 23.7.2 and do not have these issues. 

Is there something in the log? 

I briefly scanned the release notes and there no hints of changes regarding modbus. 

Maybe ask in the  HA developers in their [discord](https://www.home-assistant.io/join-chat) for help - these guys are really friendly and helpful, and they know the changes from 23.6 to 23.7 i detail

ajbatchelor, 2023-07-15:
Thanks for looking into this @mkaiser.  I'll reach out to the HA devs.  There was nothing in the logs that I could see.

I tried both 7.1 and 7.2 and on both new versions I was seeing the scan interval increase out to 20 mins.  I'll let you know if I find anything out.

mkaiser, 2023-07-16:
Lets keep the issue open. Maybe you are not the only one. Can you keep us posted?

ZEi20T, 2023-07-17:
This is also happening to me since upgrading to 2023.7 versions
[image]

also, the information in the energy dashboard looks very wrong

[image]

Louisbertelsmann, 2023-07-17:
On what kind of environment are you running Home Assistant? It could be possible that it detects being overloaded and slows down the scan interval. It shouldn't happen, but it could.

ZEi20T, 2023-07-19:
Its running on HAOS on core i5 with 16gb RAM. the sensors all appear to update fine in my dashboard. only happened after July updates

ajbatchelor, 2023-07-21:
Mines running on a RaspberryPi 4.  I've not had a chance to take this up with the HA devs on Discord, but will do so.

mkaiser, 2023-08-02:
any news on this?

btw. I am running HA on a RPI4 (4GB), too. Lukily no problems so far

Tsopic, 2023-08-04:
> This is also happening to me since upgrading to 2023.7 versions [image]
> 
> also, the information in the energy dashboard looks very wrong
> 
> [image]

@ZEi20T 
- Do you have an example where battery lvl is handled more smoothly? 

I've seen such Battery level updates since the beginning, it has never been smooth for me

ajbatchelor, 2023-08-07:
> On what kind of environment are you running Home Assistant? It could be possible that it detects being overloaded and slows down the scan interval. It shouldn't happen, but it could.

mkaiser, 2023-08-10:
[image]

my config. So add the IP of the Sungrow Inverter as "Upstreamhost"

mkaiser, 2023-08-10:
in secrets.yaml use the ip address "0" to connect with modbusprox local to HA

`sungrow_modbus_host_ip: 0`

ajbatchelor, 2023-08-13:
Hey @mkaiser, I think that I have got this working again.  I used your instructions for modbus proxy, upgraded to HA2023.8 and then removed quite a few of the sensors that I was not referencing.  It all seems to be working fine now and polling regularly.  

Thank you very much for you help!

ajbatchelor, 2023-08-21:
Thanks @mkaiser, closing this one out. One less thing on your list :)

closed 2023-08-21 by ajbatchelor

mkaiser, 2023-08-21:
One last thing I like to know: could it be caused by the bloated up amount of sensors or was the culprit most likely the general connection, which was fixed b the proxy?