mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #131: Resets to "Self consumption mode" at 1:00 + EMS self consume importing from grid not desired

Tsopic, 2023-07-20:
There are 2 problems with my plant

1) Self-consumption mode will import energy from the grid to fill the battery capacity 
 - Not desired during High PV availability season when

2) It somehow resets the EMS mode to Self-consumption at 1:00.
Which triggers the energy import from the grid.

Looking for tips/guidance how to:

1) Avoid the massive importing from Grid at 1:00
2) Limit the Self-consumption mode import from Grid

Tsopic, 2023-07-21:
Adding image for context:
[image]

As seen on the image, there was forced discharge ~22:30 

And did some unwanted charging at 1:00 + 2:09

Battery capacity is not seen in the picture, but it's still well above 30kwh charge 

Using inverter SH10RT

mkaiser, 2023-07-21:
Can you double check via isolar cloud Webinterface orlocal access via app, if the timing scheduled mode is enabled? 

It looks like a scheduled charging for me at 1am (if I remember this is the default setting). Maybe it has been accidentally enabled.

Tsopic, 2023-07-21:
Seems like you are correct, but i do not know how to remove this instruction from the inverter.

[image]

mkaiser, 2023-07-21:
these are my (never changed) values:

 
[image]

Tsopic, 2023-07-28:
Turns out i had limited access to the inverter cloud.

Using the installer/distributor level access account going forward.

mkaiser, 2023-07-30:
so is this issue solved?

Tsopic, 2023-07-31:
The 1 o clock forced charging from inverter software is gone. 
[image]
These are the current inverter settings

It's still importing some of the required load power. I don't know what the real reason is, maybe keeping in sync with grid frequency or something else. 

While I was initially expecting the plant to work from a battery when no PV energy is available. 

Currently, it uses PV when available
Supplements part of the load from GRID
- Has Backup capacity it somewhat consumes when no PV 

Closing this ticket, thanks to all participants 👍

closed 2023-07-31 by Tsopic