mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #138: Dashboard_EMS.png

svenschedlbauer, 2023-08-10:
Hello, 
How can I add the dashboard EMS? Can somebody pls explain what kind of cards are used?
Thanks in advance!!!!

mkaiser, 2023-08-11:
Did you see the instructions here?
 https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/doc/dashboard.md

svenschedlbauer, 2023-08-12:
Hello,
Thank you very much. This I haven't seen, but now I have it. Is there somewhere also an instrction for all the automatisations, because I have now the problem, that HA is changing the point "Set forced charge discharge power in W" automatically, which I don't want. Why he is chaniging it?

svenschedlbauer, 2023-08-12:
Again it is changing automatically the point "Set forced charge discharge power in W" why the Inverter is importing which is 
worse. Where is the problem? I don't understand why.... 
[image]
[image]

mkaiser, 2023-08-24:
that is strange...

do you have a load schedule defined? You need to use the app / isolarcloud (installer's access) for this

mkaiser, 2023-09-04:
Can you update to the current version? I fixed some strange stuff with the automations and string handling

closed 2023-10-10 by mkaiser