mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #141: Integration broken in new HA update

NBD99, 2023-08-18:
I just upgraded HA supervisor and after that upgrade i get invalid configuration.

Check and Restart
A basic validation of the configuration is automatically done before restarting. The basic validation ensures the YAML configuration doesn't have errors which will prevent Home Assistant or any integration from starting. It's also possible to only do the basic validation check without restarting.
Configuration invalid!
Invalid config for [modbus]: Error in sensor BDC rated power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][19]. Got {'name': 'BDC rated power', 'unique_id': 'sg_bdc_rated_power', 'slave': 1, 'address': 5627, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 100, 'scan_interval': 600}
Error in sensor BMS max. charging current swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][20]. Got {'name': 'BMS max. charging current', 'unique_id': 'sg_bms_max_charging_current', 'slave': 1, 'address': 5634, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60}
Error in sensor BMS max. discharging current swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][21]. Got {'name': 'BMS max. discharging current', 'unique_id': 'sg_bms_max_discharging_current', 'slave': 1, 'address': 5635, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60}
Error in sensor Backup phase A power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][23]. Got {'name': 'Backup phase A power', 'unique_id': 'sg_backup_phase_a_power', 'slave': 1, 'address': 5722, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}
Error in sensor Backup phase B power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][24]. Got {'name': 'Backup phase B power', 'unique_id': 'sg_backup_phase_b_power', 'slave': 1, 'address': 5723, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}
Error in sensor Backup phase C power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][25]. Got {'name': 'Backup phase C power', 'unique_id': 'sg_backup_phase_c_power', 'slave': 1, 'address': 5724, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}
Error in sensor Battery capacity swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][112]. Got {'name': 'Battery capacity', 'unique_id': 'sg_battery_capacity', 'slave': 1, 'address': 33048, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'unit_of_measurement': 'kWh', 'device_class': 'energy', 'scale': 0.01, 'scan_interval': 600}
Error in sensor Battery charging start power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][113]. Got {'name': 'Battery charging start power', 'unique_id': 'sg_battery_charging_start_power', 'slave': 1, 'address': 33148, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10}
Error in sensor Battery current swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][84]. Got {'name': 'Battery current', 'unique_id': 'sg_battery_current', 'slave': 1, 'address': 13020, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 1, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 0.1, 'scan_interval': 10}
Error in sensor Battery discharging start power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][114]. Got {'name': 'Battery discharging start power', 'unique_id': 'sg_battery_discharging_start_power', 'slave': 1, 'address': 33149, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10}
Error in sensor Battery forced charge discharge cmd raw swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][103]. Got {'name': 'Battery forced charge discharge cmd raw', 'unique_id': 'sg_battery_forced_charge_discharge_cmd_raw', 'slave': 1, 'address': 13050, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'scan_interval': 10}
Error in sensor Battery forced charge discharge power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][104]. Got {'name': 'Battery forced charge discharge power', 'unique_id': 'sg_battery_forced_charge_discharge_power', 'slave': 1, 'address': 13051, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scan_interval': 10}
Error in sensor Battery level swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][86]. Got {'name': 'Battery level', 'unique_id': 'sg_battery_level', 'slave': 1, 'address': 13022, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 1, 'unit_of_measurement': '%', 'device_class': 'battery', 'state_class': 'measurement', 'scale': 0.1, 'scan_interval': 60}
Error in sensor Battery max charge power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][110]. Got {'name': 'Battery max charge power', 'unique_id': 'sg_battery_max_charge_power', 'slave': 1, 'address': 33046, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10}
Error in sensor Battery max discharge power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][111]. Got {'name': 'Battery max discharge power', 'unique_id': 'sg_battery_max_discharge_power', 'slave': 1, 'address': 33047, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10}
Error in sensor Battery power raw swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][85]. Got {'name': 'Battery power raw', 'unique_id': 'sg_battery_power_raw', 'slave': 1, 'address': 13021, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}
Error in sensor Battery state of health swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][87]. Got {'name': 'Battery state of health', 'unique_id': 'sg_battery_state_of_health', 'slave': 1, 'address': 13023, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': '%', 'state_class': 'measurement', 'scale': 0.1, 'scan_interval': 600}
Error in sensor Battery voltage swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][83]. Got {'name': 'Battery voltage', 'unique_id': 'sg_battery_voltage', 'slave': 1, 'address': 13019, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'V', 'device_class': 'Voltage', 'scale': 0.1, 'scan_interval': 10}
[... cut ...] n': 0, 'scale': 1, 'scan_interval': 10}
Error in sensor Total backup power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][22]. Got {'name': 'Total backup power', 'slave': 1, 'unique_id': 'sg_total_backup_power', 'address': 5725, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}. (See ?, line ?).

horstvogel, 2023-08-18:
for me the same. I go from 2023.08.2 to 2023.08.3. There are some change in Home Assistant, but I don´t kown what to do.

DrBando, 2023-08-18:
same here, broken since the upgrade. Was running like a charme before. I did try to change the data types of BDC rated power but then the error just moved to the next register.

mkaiser, 2023-08-18:
big uuuff....

could you try to modify some of the "counts" and see, if the errors disappear for some registers?

I was just about to leave the computer for an extra long computer-less weekend .... Shit

DrBando, 2023-08-18:
> I tried now for one of the registers that had a count of 1 to 2 then its
> now invalid anymore.

I have changed every count 1 to 2 except for inverter and battery
temperature. The home assistant error notification is now gone and some
entities are working again, but it seems not all of them. I'll have a look
tomorrow if some more entities have recovered.

stephan192, 2023-08-19:
I had the same issue. I made a downgrade to **2023.8.2** and everything is working again.

To perform the downgrade:
1. Install the "Terminal & SSH" Addon
2. Log into your Home Assistant via SSH. I use PuTTY
3. Run the following command to downgrade: `ha core update --version 2023.8.2`

The [changelog](https://www.home-assistant.io/blog/2023/08/02/release-20238/#release-202383---august-18) states four changes for modbus in the update to 2023.8.3 and even one in the update to 2023.8.2

Louisbertelsmann, 2023-08-19:
There's also usually a partial backup made before the update. You can restore the old version by going to backups, clicking on the backup done by the update and reverting back to it. That only changes the HA version. It's a bit easier for people not comfortable in the terminal.
Also, I had the same problem. I'm gonna check if changing the counts works.

ZEi20T, 2023-08-19:
i had the problem also, tried changing the count, only some entities were working

downgrading to 8.3 has fixed the problem

Louisbertelsmann, 2023-08-19:
Changing swap: word to swap: none worked for some entities, but some are either reporting nothing or completely wrong values. Btw, that was suggested by Tyron Brandt on the Discord.

Louisbertelsmann, 2023-08-19:
I'm gonna set up modbus proxy and a new HA instance. I'm gonna try a bit and report back. Any help is appreciated.

WishMaster666, 2023-08-19:
It should be fixed in 2023.8.4 as I read here: https://github.com/home-assistant/core/pull/97798#issuecomment-1684890907

mkaiser, 2023-08-21:
fixed in 5af1945325eefc5d360cf219a03ef6b981e3ed3e

closed 2023-08-21 by mkaiser

mega-hz, 2024-01-05:
I just upgraded HA to 
Home Assistant core 2024.1.0
Home Assistant Operating System 11.3
Home Assistant Supervisor 2023.12.0

many errors with 

swap: none,
retries: (is not supported anymore)

MKaiser:
please give us a fixed File.

Louisbertelsmann, 2024-01-06:
There is a fixed file in the repository. Just update to that.