mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #143: Bug nach Update Home Assistant Core mit Modbus 2023.8.2 auf 2023.8.3

mjunker61, 2023-08-19:
The problem

OS: Proxmox 5.15.108-1-pve

Home Assistant 2023.8.2 als VM unter Proxmox
Supervisor 2023.08.1
Operating System 10.4
Frontend 20230802.0 - latest

Home Assistant Sungrow inverter integration
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
by Martin Kaiser
last update: 2023-07-08

Nach Update des Home Assistant Core 2023.8.2 auf 2023.8.3 kann der Wechselrichter Sungrow SH10RT nicht mehr über die Modbus Integration angesprochen werden. Es hagelt Fehlermeldungen. Das ist reproduzierbar. Es muss irgendetwas umbenannt worden sein, sodaß die Variablen nicht mehr angesprochen werden.

Nach Zurückspielen des Backups von gestern, geht alles wieder.
What version of Home Assistant Core has the issue?

Home Assistant Core 2023.8.3
What was the last working version of Home Assistant Core?

Home Assistant Core 2023.8.2
What type of installation are you running?

Home Assistant OS
Integration causing the issue

Modbus via configuration.yaml
Link to integration documentation on our website

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
Diagnostics information

No response
Example YAML snippet

No response
Anything in the logs that might be useful for us?

No response
Additional information

No response

nixis, 2023-08-19:
After Update to the newest HA Version, i get an erro message (below), at the HA community site there is a topic for the same problem too.

https://community.home-assistant.io/t/sungrow-modbus-integration-does-not-work-since-the-last-core-update/604560

Hope someone can help

`Configuration invalid!
Invalid config for [modbus]: Error in sensor BDC rated power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][19]. Got {'name': 'BDC rated power', 'unique_id': 'sg_bdc_rated_power', 'slave': 1, 'address': 5627, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 100, 'scan_interval': 600}
Error in sensor BMS max. charging current swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][20]. Got {'name': 'BMS max. charging current', 'unique_id': 'sg_bms_max_charging_current', 'slave': 1, 'address': 5634, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60}
Error in sensor BMS max. discharging current swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][21]. Got {'name': 'BMS max. discharging current', 'unique_id': 'sg_bms_max_discharging_current', 'slave': 1, 'address': 5635, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60}
Error in sensor Backup phase A power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][23]. Got {'name': 'Backup phase A power', 'unique_id': 'sg_backup_phase_a_power', 'slave': 1, 'address': 5722, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}
Error in sensor Backup phase B power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][24]. Got {'name': 'Backup phase B power', 'unique_id': 'sg_backup_phase_b_power', 'slave': 1, 'address': 5723, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}
Error in sensor Backup phase C power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][25]. Got {'name': 'Backup phase C power', 'unique_id': 'sg_backup_phase_c_power', 'slave': 1, 'address': 5724, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}
Error in sensor Battery capacity swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][112]. Got {'name': 'Battery capacity', 'unique_id': 'sg_battery_capacity', 'slave': 1, 'address': 33048, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'unit_of_measurement': 'kWh', 'device_class': 'energy', 'scale': 0.01, 'scan_interval': 600}
Error in sensor Battery charging start power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][113]. Got {'name': 'Battery charging start power', 'unique_id': 'sg_battery_charging_start_power', 'slave': 1, 'address': 33148, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10}
Error in sensor Battery current swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][84]. Got {'name': 'Battery current', 'unique_id': 'sg_battery_current', 'slave': 1, 'address': 13020, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 1, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 0.1, 'scan_interval': 10}
Error in sensor Battery discharging start power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][114]. Got {'name': 'Battery discharging start power', 'unique_id': 'sg_battery_discharging_start_power', 'slave': 1, 'address': 33149, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10}
Error in sensor Battery forced charge discharge cmd raw swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][103]. Got {'name': 'Battery forced charge discharge cmd raw', 'unique_id': 'sg_battery_forced_charge_discharge_cmd_raw', 'slave': 1, 'address': 13050, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'word', 'precision': 0, 'scan_interval': 10}
Error in sensor Battery forced charge discharge power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][104]. Got {'name': 'Battery forced charge discharge power', 'unique_id': 'sg_battery_forced_charge_discharge_power', 'slave': 1, 'address': 13051, 'input_type': 'holding', 'co [... cut ...] ': 0, 'scale': 1, 'scan_interval': 10}
Error in sensor Total backup power swap(word) not possible due to the registers count: 1, needed: 2 @ data['modbus'][0]['sensors'][22]. Got {'name': 'Total backup power', 'slave': 1, 'unique_id': 'sg_total_backup_power', 'address': 5725, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'word', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10}. (See ?, line ?).`

smartcuc, 2023-08-19:
Same issue after update on end too. However, still believe it's relate to HA and not the intigration.

nixis, 2023-08-19:
Should be fixed with 2023.8.4, see here:
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/141#issuecomment-1684894823

mkaiser, 2023-08-21:
fixed in 5af1945325eefc5d360cf219a03ef6b981e3ed3e

closed 2023-08-21 by mkaiser