mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #146: load_power sensor only shows positive load

koch86, 2023-08-24:
I am wondering if there is an possibility to sense negative values for the load_power sensor.
In my setup, there is a separate micro inverter (Hoymiles) connected to a plug in my house (before the smart meter). This leads to energy export in sunny hours and therefor negative load values.

But: the sensor load_power cuts off any negative load and therefor it is not possible to measure/calculate the real power consumption of the house.

Question: Can this value be obtained somehow? Tried different values and also some calculations but never found the value I am looking for… :-(

[image]

dylan09, 2023-08-25:
Maybe you could have a look at `sensor.export_power_raw`. Should be negative when importing power. But don't know if its possible to calculate the value you wish to get?

koch86, 2023-08-25:
> Maybe you could have a look at `sensor.export_power_raw`. Should be negative when importing power. But don't know if its possible to calculate the value you wish to get?

Unfortunately this sensor does not help in my case…

just for clarification:

House consumes 250 W (unknown value)
Micro Inverter generates 400 W (known value)
Sungrow SmartMeter real input is -150 W but `sensor.load_power` measures 0 W
Sungrow Inverter generates `sensor.total_dc_power` 5000 W
Export `sensor.export_power_raw` (and also `sensor_export_power`) is 5000 W

I would like to calc the house consumption so I need the actual load of the SmartMeter input to calc the houses power consumption.

dylan09, 2023-08-25:
Then maybe the value of `sg_meter_active_power` could be the right value. It's not officially documented. But could be the real meter value?

koch86, 2023-08-26:
Unfortunately not - even the difference between real meter load an DC power is not the value I am looking for.
Decided to install another Shelly 3EM to monitor the load.

Thanks for your assistance!

closed 2023-08-26 by koch86

elektrinis, 2023-08-27:
I have a second inverter connected per sungrow manual, and I am experiencing this non-negative measurement as well. I believe it's a bug in sungrow firmware.