mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #148: Problem with config.yaml

Ruebnow, 2023-08-25:
Hi, after making the entry in config.yaml, I get the following error when checking: "Error loading /config/configuration.yaml: could not determine a constructor for the tag '!include_modbus_sungrow.yaml'
in "/config/configuration.yaml", line 8, column 13" . Can you help me? The entry in config.yaml looks like this "
  packages: !include_modbus_sungrow.yaml"

mkaiser, 2023-08-25:
My first guess it, that it coult be related to indentation, try this (put the modbus_sungrow.yaml in a folder "packages" first)

```
homeassistant:
  packages:
    sungrow: !include packages/modbus_sungrow.yaml
```

"homeassistant:" has no identaion, for "packages" there are 2 spaces, for "sungrow" 4 spaces

Ruebnow, 2023-08-26:
> <mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/148/1693852159@
> github.com>

hi, first of all thanks for the answer. It was my fault because I
misunderstood something. after i found my error also my inverter was
running in ha. thanks

closed 2023-08-26 by Ruebnow