mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #154: Battery percentage charge spikes magically to ~30%

arfoll, 2023-08-30:
I have an SH5.0RT and since the HA 8.4 update & corresponding fixes from here I have noticed weird battery charge/discharge levels. It seems it jumps to 30% as soon as it starts charging and this dissapears again when you get to 30% :) Small screenshot showing the issue - I have a 10KW battery it was at 0 this morning and so far I've generated 0.8kW so clearly the battery cannot be this high (or have charged this quickly :)).

[image]

I haven't really looked at what changed much but I didn't change anything on the inverter or setup so I can only imagine the error comes from this but it's possible the error existed for a while and I didn't notice, it was very sunny so it's possible I would have missed an erroneous 30% since I never went to 0%.

mkaiser, 2023-09-02:
[image]

what are your SoC / reserved settings?

arfoll, 2023-09-02:
I've never touched them :) so they are completely default:
5%
100%
0%

Tbh I've no idea what I should use...

Fyi seems it's back to normal in the last few days, I reconnected the cloud connection to double check the stats and seems to line up again...

mkaiser, 2023-09-11:
okay, please create a new thread, if this happens again :)

closed 2023-09-11 by mkaiser