mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #155: Update the code for home assistant 2023.8.4

bows33r, 2023-09-05:
Hi! Can anyone tell me what in my code i have to change to get my intergration to work again? 

# Home Assistant Sungrow inverter integration
# https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
# by Martin Kaiser
# 13th Dezember 2022

modbus:
  - name: SungrowSG20RT
    type: tcp
    host: !secret sungrow_modbus_host_ip
    port: !secret sungrow_modbus_port
    retry_on_empty: true
    # retries: 10
    # close_comm_on_error: true
    delay: 5
    #timeout: 5
    sensors:
      # My own SG20RT setup begin

      - name: Solceller installerad effekt
        slave: !secret sungrow_modbus_slave
        address: 5000
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 1
        unit_of_measurement: kW
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 10

      - name: Solceller produktion dag
        slave: !secret sungrow_modbus_slave
        address: 5002
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 1
        unit_of_measurement: kWh
        device_class: energy
        state_class: total_increasing
        scale: 0.1
        scan_interval: 10

      - name: Solceller invertertemperatur
        slave: !secret sungrow_modbus_slave
        address: 5007
        input_type: input
        count: 1
        data_type: int16
        precision: 1
        unit_of_measurement: °C
        device_class: Temperature
        scale: 0.1
        scan_interval: 10

      - name: Solceller total skenbar effekt
        slave: !secret sungrow_modbus_slave
        address: 5008
        input_type: input
        count: 2
        data_type: uint32
        swap: word
        precision: 0
        unit_of_measurement: W
        device_class: power
[... 363 lines cut ...]
      - name: Solceller slinga 1 ström
        slave: !secret sungrow_modbus_slave
        address: 7012
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 2
        unit_of_measurement: A
        device_class: Current
        scale: 0.01
        scan_interval: 10

      - name: Solceller slinga 2 ström
        slave: !secret sungrow_modbus_slave
        address: 7014
        input_type: input
        count: 1
        data_type: uint16
        swap: word
        precision: 2
        unit_of_measurement: A
        device_class: Current
        scale: 0.01
        scan_interval: 10

mkaiser, 2023-09-06:
this is not an issue with the current code base. 

You will have to compare your code with the git. take a look at the "swap" keywords. 

If you want a quick and easy way to just translate the entities, use the current code from the git repo and use the home assistant "customize" approach.

I'd will be happy to include some Scandinavian translations, if you woul like to contribute

bows33r, 2023-09-06:
> this is not an issue with the current code base.
>
> You will have to compare your code with the git. take a look at the "swap" keywords.
>
> If you want a quick and easy way to just translate the [...]

Thank you! i got it working again! But can you please explain what the "swap word or none" means? 

Sincerely!