mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #156: Index out of range and Input Select errors

mrebbert, 2023-09-06:
Hi,

i got an SH6.0RT-V112 since yesterday.
After integration of the package and restarting everything i get a bunch of recurring errors (see below) regarding:
- Index out of range
- input select

I already turn on the debug mode for modbus, this is the detailed output i get.

Do you have a clue?

KR,
Markus

home-assistant  | 2023-09-06 09:56:13.460 ERROR (SyncWorker_8) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.461 ERROR (SyncWorker_8) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.462 ERROR (SyncWorker_8) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.476 ERROR (SyncWorker_7) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.477 ERROR (SyncWorker_7) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.491 ERROR (SyncWorker_1) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.491 ERROR (SyncWorker_1) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.507 ERROR (SyncWorker_2) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.508 ERROR (SyncWorker_2) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.523 ERROR (SyncWorker_6) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.524 ERROR (SyncWorker_6) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.539 ERROR (SyncWorker_11) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.540 ERROR (SyncWorker_11) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.555 ERROR (SyncWorker_8) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.555 ERROR (SyncWorker_8) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.570 ERROR (SyncWorker_1) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.571 ERROR (SyncWorker_1) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.655 ERROR (SyncWorker_6) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.656 ERROR (SyncWorker_6) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.657 ERROR (SyncWorker_6) [homeassistant.components.modbus.modbus] Pymodbus: SungrowSHx: Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.680 ERROR (SyncWorker_11) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.682 ERROR (SyncWorker_11) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.700 ERROR (SyncWorker_5) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.701 ERROR (SyncWorker_5) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.706 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_cmd_input_select_update] sungrow inverter update battery forced charge discharge cmd input select update: Error executing script. Error for call_service at pos 1: Invalid option: Unknown - should not see me! (possible options: Stop (default), Forced charge, Forced discharge)
home-assistant  | 2023-09-06 09:56:13.712 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_cmd_input_select_update] Error while executing automation automation.sungrow_inverter_update_battery_forced_charge_discharge_cmd_input_select_update: Invalid option: Unknown - should not see me! (possible options: Stop (default), Forced charge, Forced discharge)
home-assistant  | 2023-09-06 09:56:13.720 ERROR (SyncWorker_9) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.721 ERROR (SyncWorker_9) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.724 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update] sungrow inverter update battery forced charge discharge power input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
home-assistant  | 2023-09-06 09:56:13.727 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update] Error while executing automation automation.sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update: expected float for dictionary value @ data['value']
home-assistant  | 2023-09-06 09:56:13.740 ERROR (SyncWorker_4) [pymodbus.logging] General exception: index out of range
home-assistant  | 2023-09-06 09:56:13.741 ERROR (SyncWorker_4) [pymodbus.logging] Modbus IO exception Modbus Error: [Input/Output] Unable to decode request
home-assistant  | 2023-09-06 09:56:13.745 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_max_soc_input_slider_update] sungrow inverter max SoC input slider update: Error executing script. Invalid data for call_service at pos 1: expected float for dictionary value @ data['value']
home-assistant  | 2023-09-06 09:56:13.748 ERROR (MainThread) [homeassistant.components.automation.sungrow_inverter_max_soc_input_slider_update] Error while executing automation automation.sungrow_inverter_max_soc_input_slider_update: expected float for dictionary value @ data['value']

mrebbert, 2023-09-06:
Sorry, my bad. Saw, that the LAN Port is actually not used. I'll use the LAN port instead and try it again.

closed 2023-09-06 by mrebbert

mkaiser, 2023-09-06:
Also make sure, that the inverters are upgraded to the newest firmware :)

mrebbert, 2023-09-06:
it is :)