mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #158: Exceptions since HA 2023.9.0

ndurchx, 2023-09-08:
Since HA 2023.9.0 I get exceptions:
```
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 591, in state
    numerical_value = float(value)  # type:ignore[arg-type]
                      ^^^^^^^^^^^^
ValueError: could not convert string to float: 'unavailable'
...
ValueError: Sensor sensor.monthly_pv_generation_current has device class 'energy', state class 'total_increasing' unit 'kWh' and suggested precision 'None' thus indicating it has a numeric value; however, it has the non-numeric value: 'unavailable' (<class 'str'>)
```
Afterwards I have to reload the yaml to bring the sensor values back. Perhaps this is caused by an unstable lan connection. Would be a solution to check every value for the string "unavailable" before using it as sensor value?

mkaiser, 2023-09-08:
yes, you are right. 

When the yaml is loaded, some modbus values are still not present. The template sensors should check this before.

mjunker61, 2023-09-08:
The same here, I have no sensors. Is it the same bug, as to update 2023.8.3 ?
What is the crew from homeassistant doing ?

mkaiser, 2023-09-08:
well its more broken in my code :)

But I don't know yet, why it is triggering now

Louisbertelsmann, 2023-09-08:
an "availability" option in the templates could solve it, I can add those if you want. Of course only if you aren't working on the code a lot, so I don't trigger too many conflicts merging.

mjunker61, 2023-09-08:
Above I forgot to say, that I make a downgrade to core-2023.8.4 and now it works for now.

mkaiser, 2023-09-08:
Can you double check, if it is fixed?

I am on 2023.09 and there are no more log entries after "full" restart  and none after "yaml reload"

ndurchx, 2023-09-09:
The error does not come up again. 2023.09 and 2023.09.01
The sensors do not work like before ha update, but i think it is more a problem of the modbus integration itself: https://github.com/home-assistant/core/issues/99784

HKUser1, 2023-09-09:
Hello Same here no Sensors again since 2323.09.01.
If I do a restart it woks for e few seconds or some minutes.... then the Sensors are offline again..

HKUser1, 2023-09-10:
Anything new about the Modbus Problem since 23.09.0 I can not use my surplus charging :)

mjunker61, 2023-09-10:
> Can you double check, if it is fixed?
> 
> I am on 2023.09 and there are no more log entries after "full" restart and none after "yaml reload"

The modbus_sungrow.yaml last update: 2023-09-08 (2) doesn't work here. 

I made for testing and logging a snapshot from my working ha-vm with ha core-2023-08.4. Than I updated the modbus_sungrow.yaml  and after that I updated the ha core. After Update to ha core-2023-09.1 the sensors works for maybe 20 minutes, then sensores are not available. HA works completely a bit slower (not testing, only feeling by clicking ;)) 

Here is my abstracted Log with ha core-2023-09.1 and Sungrow Hybrid Inverter SH10RT:

[auszug-log-nach-core-update-2023-09.1-home-assistant_2023-09-10T18-06-19.729Z.log](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/12569957/auszug-log-nach-core-update-2023-09.1-home-assistant_2023-09-10T18-06-19.729Z.log)

TheFreaker86, 2023-09-10:
The update 2023-09-08 (2) works well with the SH10RT. The previous version worked as well until something broke. So thank you very much for the quick fix!

storkio, 2023-09-10:
Hello , same here since 2323.09.01. 
modbus_sungrow.yaml update 2023-09-08 (2) doesn't help.
After restart or yaml reload it woks for a few minutes and goes offline again.

TheFreaker86, 2023-09-11:
I reverted to Home Assistant version 2023.8.4. The more recent HA release doesn't work

HKUser1, 2023-09-11:
i tried two times to resore 2023.08.4 but it doesn t work. What am i doing wrong.
HA disappears and only a reboot after 30 min brings HA back with 2023.9.1

mkaiser, 2023-09-11:
Who of you is using a modbus proxy?

I am using one and not experiencing all this issue. 

Could you maybe try https://github.com/Akulatraxas/ha-modbusproxy/ ?

TheFreaker86, 2023-09-11:
> Who of you is using a modbus proxy?
> 
> I am using one and not experiencing all these issues.
> 
> Could you maybe try https://github.com/Akulatraxas/ha-modbusproxy/ ?

I don't. At least I did not configure one.

HKUser1, 2023-09-12:
i would dongradw till the Problem is fixed but i can`t do it.
HA ist not starting after restoring the Backup...Then i disconnect from power then HA starts with 2023.09 again..

hmmm

BrettLynch123, 2023-09-13:
FYI seems to be fixed in HA 2023.9.2 and fixed by this PR https://github.com/home-assistant/core/pull/99988
Has been working for 5 minutes so far for me. Would usually fail by now. Will update this if it stops working for me.

HKUser1, 2023-09-13:
with HA 2023.9.2 everything is working fine since this morning.

mkaiser, 2023-09-13:
> FYI seems to be fixed in HA 2023.9.2 and fixed by this PR [home-assistant/core#99988](https://github.com/home-assistant/core/pull/99988) ~Havent been able to test properly though.~ Edit: Seems to be working ok for 1.5 hours ok so far. but data looks weird. Seems to have breaks in the data so maybe not fixed but less bad.
> 
> What it should be [image]
> 
> What I'm getting [image]

sorry, I cannot derive anything from the plots. What is shown there? (Axis description is missing)

TheFreaker86, 2023-09-14:
works fine with the latest HA release. Thank you very much! 👍

closed 2023-10-10 by mkaiser