mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #159: Battery Sensor (W)

mjunker61, 2023-09-08:
In the file "modbus_sungrow.yaml" should the power with kW configured, like below. Otherwise the Sensor cannot show the correct Value in the GUI. 

I did it, and now I have the correct Value in the Dashboard.

- name: Battery charging power # positive if charging else zero
        ...
        unit_of_measurement: kW
        ...

      - name: Battery discharging power # positive if discharging else zero
        ...
        unit_of_measurement: kW

mkaiser, 2023-09-08:
What inverter are you using?

My SH10RT shows the corrrect values  (in W, not kW)
[image]

SieresGH, 2023-09-09:
Hello, I wanted to get involved. This is displayed correctly for me. The problem, however, is that neither "Battery charging power" or "Battery discharging power" is displayed correctly. For me the number remains at zero, see screenshot. Battery power raw is displayed normally, as well as many others are also displayed correctly. I have a Sungrow SH10RT inverter and BYD storage. I use HomeAssistant version 2023.8.4.
[image]

mjunker61, 2023-09-10:
We have a Sungrow Hybrid Inverter SH10RT. Maybe the reason was, that it was one of the first SH10RT, sold in GE. The first steps were very stressfull with a lot of communication, Resets and Firmware Updates. Now it works good. 

Here my screenshots for you:

[image]

[image]

I didn't check all sensors, but I think it is the same at meter_active_power and so on here.

mkaiser, 2023-09-11:
So this issue can be closed? :)

closed 2023-10-10 by mkaiser

Louisbertelsmann, 2023-10-11:
No, it doesn't. Some people prefer it to be in W, some in kW. You can set it to anything by just changing the unit of measurement and the scale. For me, this config works and shows the correct values.

mjunker61, 2023-10-12:
@Louisbertelsmann and @mkaiser 
Pls, I do not want to waste your all time. I only want to inform and say, in my homeassistant I saw a problem and I have to declare it manualy = "unit of measurement" in "kW", otherwise I get the wrong values. 

It has nothing to do with formatting it in any guis or so. Certainly I checked this, befor I wrote this Issue.

But when I am the only one, who have this ==> it is no problem. I change it manualy an all is ok. 
I only want to help !!!

Louisbertelsmann, 2023-10-12:
What you could try is doing a Firmware Update. If you already created an installer Account, so it that way. If not, so it via the App and local access.

dylan09, 2023-10-12:
@mjunker61 I think there is an error in your configuration or something went wrong inside your inverter.

Sungrow protocol definition states `1W` as unit for register 13022.
Maybe you could check the definition for `sensor.battery_power_raw` in your configuration. Is it the same as in github?