mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #161: Enable/Disble Off Grid option (feature request)

optimismus, 2023-09-12:
**Is your feature request related to a problem? Please describe.**
I would like to set the off grid function via HA.

**Describe the solution you'd like**
Implement the modbus registers. "Scottytauber" says this is to do:
"the Holding Register should be 13074 Off Grid option on the Sungrow SH10RT.
0xAA = On
0x55 = Off"

**Describe alternatives you've considered**
You don't implement :-)

**Additional context**
I already asked for this under discussion, you asked me to file an issue.

Kind regards
Axel

tttooommmeeekkk, 2024-04-02:
Hey,

I also had this request and by now found this "feature request".
Meanwhile I added some code to my local modbus_sungrow.yaml
Works fine for me (I can hear the loud "klack" inside the inverter when switching).
**### But be aware (mine HA-Version is not the actual one - I use HA-core 2023.11.2)! ###**

Here are the code-snippets for a possible feature-release:

### modbus - sensors -part:
```
      - name: offgrid option raw
        unique_id: sg_offgrid_option_raw
        device_address: !secret sungrow_modbus_slave
        address: 13074 # reg 13075
        input_type: holding
        data_type: uint16
        precision: 0
        state_class: measurement
        scan_interval: 31
```

### template sensor -part:
```
- name: offgrid option
        unique_id: sg_offgrid_option
        availability: "{{ not is_state('sensor.offgrid_option_raw', 'unavailable') }}"
        # TODO: test state_class with enum
        # state_class: measurement
        device_class: enum
        state: >-
          {% if ((states('sensor.offgrid_option_raw') | int(default=0)) == 0x00AA) %}
            Enabled
          {% elif ((states('sensor.offgrid_option_raw') | int(default=0))  == 0x0055) %}
            Disabled
          {% else %}
            Unknown - should not see me!
          {% endif %}
```
### input_select -part:
```
  set_sg_offgrid_option:
    name: offgrid option
    options:
      - "Enabled"
      - "Disabled"
    icon: mdi:power-standby
```

### automation -part:
```
  - id: "automation_sungrow_inverter_offgrid_option_update"
    alias: "sungrow inverter offgrid option update"
    description: "Updates Enable/Disable for offgrid option"
    trigger:
      - platform: state
        entity_id:
          - sensor.offgrid_option_raw
    condition:
      - condition: template
        value_template: "{{ not is_state('sensor.offgrid_option_raw', 'unavailable') }}"
    action:
      - service: input_select.select_option
        target:
          entity_id: input_select.set_sg_offgrid_option
        data:
          option: >
            {% if ((states('sensor.offgrid_option_raw') | int(default=0)) == 0x00AA) %} 
              Enabled
            {% elif ((states('sensor.offgrid_option_raw') | int(default=0)) == 0x0055) %} 
              Disabled
            {% endif %}
    mode: single

  - id: "automation_sungrow_offgrid_option"
    alias: "sungrow inverter offgrid option"
    description: "Set offgrid option Ersatzstromfunktion"
    trigger:
      - platform: state
        entity_id:
          - input_select.set_sg_offgrid_option
    condition: []
    variables:
      offgrid_option_enable: 0xAA
      offgrid_option_disable: 0x55
    action:
      - service: modbus.write_register
        data_template:
          hub: SungrowSHx
          slave: !secret sungrow_modbus_slave
          address: 13074 # reg 13075
          value: >
            {% if is_state('input_select.set_sg_offgrid_option', "Enabled") %} 
              {{offgrid_option_enable}}
            {% elif is_state('input_select.set_sg_offgrid_option', "Disabled") %}
              {{offgrid_option_disable}}
            {% else %}
              {{offgrid_option_disable}}
            {% endif %}
    mode: single
``` 

Finally it looks like this (red marked area):
[image]

Louisbertelsmann, 2024-04-05:
I will try implementing.

elektrinis, 2024-04-05:
This looks very cool. Please report who will test it. And then I'd love to see this merged in to main.

elektrinis, 2024-04-16:
Just tried it. Latest HAOS, but I'm running old yaml.
By default register 13074 had value 0xAA. Written 0x55, but nothing changed :( Did it while sun was shining, producing more than consuming, battery half-empty. Was expecting it to disconnect from grid and charge the battery.
Do I need to do something extra, like change operating modes? Tried it in self-consumption and in Forced mode.

elektrinis, 2024-04-16:
I think I did not fully understand what it does.
There is an internal relay between AC-input and AC-backup. In my case, the whole house is connected to AC-backup.
When grid fails, inverter switches off the relay automatically to no reverse-feed in to the grid, and the inverter supplies AC-backup port from battery.
What I need is to control the internal relay to disconnect from grid when I want to, as if grid power was lost.

I don't really understand now what this register does. You said you can head a loud click, which means mechanical relay is switching. However, in my case it's silent. :/

Where did you find the info about this register? Maybe there are others that could do what I want.

Gnarfoz, 2024-04-16:
The latest Modbus register documentation from Sungrow lists this register. (Don't expect much insight, though, it just says "off-grid option".)
What this does is it enables or disables the backup power option, i.e. the "automatic switchover" which you describe.
If you set this to off, it won't automatically switch to backup power.
You can't use it to switch to backup mode at will while the grid is present. For that, you need to actually disconnect the grid.

tttooommmeeekkk, 2024-04-16:
@elektrinis 

The schematic diagram of the SHxxRT shows, that there are two AC-relays inside the device - one between the grid-port and AC/DC-transformer and the other for the backup-port and the AC/DC-transformer.
Between grid-port and backup-port there exists a bypass-relay for supply the backup-connected devices with grid-energy, when there ist no(t enough) solar- and/or battery-power.
What the here mentioned modbus-register does is to switch the AC-relay of the backup-port. As far as I understood the sungrow-installers-presentation the bypass-relais is an automatic switch that cannot be switched or read by register.
The hyperlink to the sungrow-protocol-documentation with refering register can be found here for example:
[SG-Hybrid-Modbus-Documentation V1.1.2](https://www.photovoltaikforum.com/thread/166134-daten-lesen-vom-sungrow-wechselrichtern-modbus/?postID=3452929#post3452929)

The schematic-diagram can be found here:
[schematic-diagram-extract-en](https://www.photovoltaikforum.com/core/attachment/288831-sh-intern-jpg/) or here
[schematic-diagram-de](https://www.photovoltaikforum.com/thread/141840-sungrow-sh10rt-inself%C3%A4hig/?pageNo=36)

elektrinis, 2024-04-17:
Thank you both, I think I now get it. It's not what I am looking for. Probably will have to install an external relay. Yet another re-do of the electrical panel...

### Just FYI why I need it. 
**The problem**: our energy supplier counts each phase separately, for example, when in self-consumption mode and inverter is trying to keep grid power at 0W (with battery), there is actually an imbalance, and one phase may read +100W and two phases -50W each. We have 0W overall, however the supplier reads it as if I am sending 100W to the grid and simultaneously buying back 100W with a 32% markup. This is called arithmetic metering and is not fair, but we have what we have. Most other countries have vector metering, counting only the sum.
**The solution**: to bypass this issue, I want to fully disconnect from grid when I'm on battery power. I would connect back only when my consumption is over 10kW (charging EVs etc.) or when battery is fully depleted or is fully charged and I have a surplus. I can automate this inside home assistant, by using, for example, inverter's single programmable port.
I hope the idea is clear.

Gnarfoz, 2024-04-17:
Nothing including the word "overvoltage" sounds like a good idea. 🤐
Arithmetic metering seems like a bad deal indeed.

closed 2024-12-15 by github-actions[bot]