mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #162: Forced Charge Not Working

Crootie, 2023-09-13:
**Describe the bug**
When i change the EMS to forced mode for charging it is not activating. Everything says forced but it won't forced charge.

**Your Sungrow inverter:**
 - Model: [e.g. SH-5.RT v112]
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Modbus Proxy (mark one)
  - [] yes
  - [] no

**Home Assistant :**
 - Version: [e.g. 2023.09.2]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file)
# last update: 2023-09-12

**Expected behavior**
Forced Charge should enable buut it isn't.

mkaiser, 2023-09-13:
hi, 

There was an issue with the EMS mode setting, which was fixed last night. Please ensure, that you are on the 2023-09-12 version. 

If you are on the current version: Did you set both parameters?
1) EMSmode     --> "Forced mode"
2) Battery forced charge discharge cmd --> "Forced charge"

mkaiser, 2023-09-13:
[image]

Crootie, 2023-09-13:
[image]
[image]
[image]

Yep. I updated the Code before I updated HA to 2023.09.02

mkaiser, 2023-09-13:
just to be sure: did you wait >10 seconds after setting the mode? It takes some time until the sensors are updated again:

[image]

is something in the log files?

Crootie, 2023-09-13:
have managed to get inverter state to change but it still isn't charging the battery.
Have waited about 1 minute before taking screen shots. Not sure where to go to find the logs for this but there is nothing under core or supervisor which point to anything remotely related to the pv.

[image]
[image]

Crootie, 2023-09-13:
Don't know if this is going to make a difference and after testing it hasn't. says enabled but should be disabled.

[image]

mkaiser, 2023-09-13:
Does something happen, if you toggle the "Export power limit mode"?

Also, I recommend completely powering off your system and turn it back on. I had it a couple of times, that the battery was offline.  Cold-Restarting helped.

Crootie, 2023-09-13:
Does something happen, if you toggle the "Export power limit mode"?

toggled this and it is now disabled. pulled the power on the rpi4 and left it for 5 mins then rebooted. still won't go into forced charge mode even though it says it is. can you guide me to the logs for it?

Also, I recommend completely powering off your system and turn it back on. I had it a couple of times, that the battery was offline. Cold-Restarting helped.

mkaiser, 2023-09-13:
> Does something happen, if you toggle the "Export power limit mode"?
> 
> Also, I recommend completely powering off your system and turn it back on. I had it a couple of times, that the battery was offline. Cold-Restarting helped.

oh, that was ambiguous ....

I meant: Please completely power off your inverter / battery :)

(I never cold-restarted my RPI4 the last 3 years, except one power outage...)

mkaiser, 2023-09-13:
> toggled this and it is now disabled. pulled the power on the rpi4 and left it for 5 mins then rebooted. still won't go into forced charge mode even though it says it is. can you guide me to the logs for it?

Settings --> System --> Logs

Crootie, 2023-09-13:
Found an error...

Logger: homeassistant.components.modbus.modbus
Source: components/modbus/modbus.py:401
Integration: Modbus (documentation, issues)
First occurred: 7:15:42 PM (1 occurrences)
Last logged: 7:15:42 PM

Pymodbus: SungrowSHx: Exception Response(134, 6, SlaveFailure)

mkaiser, 2023-09-13:
Can you enable the debug log for modbus and give me a new log?

Instructions here: 
https://www.home-assistant.io/integrations/modbus/

search for "logger"

Crootie, 2023-09-14:
[home-assistant_2023-09-14T09-14-31.021Z.log](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/12607065/home-assistant_2023-09-14T09-14-31.021Z.log)

okies.. try that.

mkaiser, 2023-09-14:
Cannot find any error related to modbus in your logs. Seems all working normal

Crootie, 2023-09-14:
Same but it still won't force charge.

mkaiser, 2023-09-14:
Let's go some steps back....

What happens if you are using the sungrow App for this?

Crootie, 2023-09-14:
> Let's go some steps back....
> 
> What happens if you are using the sungrow App for this?

Well this looks interesting.... not force charging through the app either. I think i might reboot the inverter tomorrow and see if that fixes it.

Crootie, 2023-09-19:
ok restarted the system multiple times. going to contact the supplier of the unit since it isn't working through the sungrow app either. fly out to work tomorrow for 2 weeks so will have limited updates. HA addon still isn't working either even though it is direct connect.

STL2020, 2023-10-01:
Same Problem here. Switch to Force mode but nothings happens :-(
Sungrow Firware latest...

mkaiser, 2023-10-02:
> Same Problem here. Switch to Force mode but nothings happens :-( Sungrow Firware latest...

see some post earlier: 
> Let's go some steps back....
> 
> What happens if you are using the sungrow App for this?

STL2020, 2023-10-03:
Yes thank you , it is a Sungrow Inverter Update Problem Maybe after the last Update  !

hAMMERlized, 2023-10-15:
Here it is either not working. Same as stated about. Forced mode seems to be on but it is not loading the battery from grid.

STL2020, 2023-10-17:
Yes I don't know why. But the same problem no update and no response from Sungrow !  I can't use the force mode maybe since the last Update  !

hAMMERlized, 2023-10-27:
> > Same Problem here. Switch to Force mode but nothings happens :-( Sungrow Firware latest...
> 
> see some post earlier:
> 
> > Let's go some steps back....
> > What happens if you are using the sungrow App for this?

i have installed the latest [hybrid inverter firmware](https://github.com/sungrow-firmware/firmware/tree/main/Hybrid%20Inverter/SH8.0RT-V122) (SAPPHIRE-H_B001.V000.P004-20230719.zip) and [communication module firmware](https://github.com/sungrow-firmware/firmware/tree/main/Communication%20Module/WiNet-S) (WiNet-S/WINET_B000.V020.P001.zip).

Battery not updated yet to the latest firmware.

Using the isolarcloud app the force charging works as expected. Using the modbus integration, EMS Mode = Forced Mode & Battery Force Charge Discharge cmd = Forced Charge, does not work. 

Inverter States "EMS mode selection" & "Battery forced charge discharge cmd" do change to force mode but unfortunately nothing happens.

By the way the "Inverter mode" drop down menu is greyed out and i can't change anything. Is this correct?

Thanks.

Hoellenwesen, 2023-11-29:
Hi,

i have also the issue that the force charge is not working. I have the latest versions installed but nothing happens when i chnage to force charge.

[image]

I waited about 5 to 10 minutes after each state switch and then about 15 minutes for the battery to begin charging.

hAMMERlized, 2023-11-29:
> Hi,
> 
> 
> 
> i have also the issue that the force charge is not working. I have the latest versions installed but nothing happens when i chnage to force charge.
> 
> 
> 
> [image]
> 
> 
> 
> I waited about 5 to 10 minutes after each state switch and then about 15 minutes for the battery to begin charging.

Seems you don't have the latest updates installed. See my comment above.

hAMMERlized, 2023-12-06:
Did you set a value at "set forced charge discharge Power"?

It also takes a few minutes until the inverter is switching the state to forced mode. 2-3 minutes in my case.

Battlier, 2023-12-07:
After one more reboot it works again.

shelldor, 2023-12-10:
I am also unable to set force mode, found this in the log.

Can anyone tell me how to set force mode in the app?

```
`Logger: homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update
Source: components/automation/__init__.py:676
Integration: Automation (documentation, issues)
First occurred: 2:51:32 AM (2 occurrences)
Last logged: 8:08:20 AM

Error while executing automation automation.sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update: expected float for dictionary value @ data['value']`
```

Why is capacity 0?
[image]

Louisbertelsmann, 2023-12-10:
Try updating to the newest version of modbus_sungrow.yaml

bliblabl8, 2023-12-24:
> I am also unable to set force mode, found this in the log.
> 
> ```
> `Logger: homeassistant.components.automation.sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update
> Source: components/automation/__init__.py:676
> Integration: Automation (documentation, issues)
> First occurred: 2:51:32 AM (2 occurrences)
> Last logged: 8:08:20 AM
> 
> Error while executing automation automation.sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update: expected float for dictionary value @ data['value']`
> ```
> 

I had the same issue. I added "|float" in line 2742 in latest version of sungrow_modbus.yaml and now force charging works again.

hAMMERlized, 2023-12-27:
> I had the same issue. I added "|float" in line 2742 in latest version of sungrow_modbus.yaml and now force charging works again.

In line 2742 only stands "data". Did you mean 2743? If yes, how does your code look in that line?

```
2742        data:
2743          value: "{{ states('sensor.battery_forced_charge_discharge_power') }}"
```

bliblabl8, 2023-12-27:
In my case it's line 2742 beginning with "value:..." as shown below:

        data:
          value: "{{ states('sensor.battery_forced_charge_discharge_power')|float }}"
    mode: single

mkaiser, 2023-12-27:
hmm It does not make much sense to me... Can someone enlighten me?

```battery_forced_charge_discharge_power``` is a sensor (modbus uint16) with precision 0. So a simple integer. 

You described that adding the ```|float``` conversion made force charging to work again:

```
  - id: "automation_sungrow_inverter_update_battery_forced_charge_discharge_power_input_slider_update"
    alias: "sungrow inverter update battery forced charge discharge power input slider update"
    description: "Updates battery forced charge discharge power input slider"
    trigger:
      - platform: state
        entity_id:
          - sensor.battery_forced_charge_discharge_power
    condition: []
    action:
      - service: input_number.set_value
        target:
          entity_id: input_number.set_sg_forced_charge_discharge_power
        data:
          value: "{{ states('sensor.battery_forced_charge_discharge_power') >>>>>>>>>>>>>> |float <<<<<<<<<<<<< }}"
    mode: single
```

(used >>> and <<< because I cant use **bold** formatting in a code quote)

The input sensor ```input_number.set_sg_forced_charge_discharge_power``` is also an integer. Why should a | float conversion help here?

What is your exact inverter model?

bliblabl8, 2023-12-27:
I've just read the error message "expect float" , so I tried that. Nothing else.

bliblabl8, 2024-01-02:
In the meanwhile, I changed my opinion. Forced charging does take a couple of minutes (! Up to 10 minutes) to start when converter isn't active (e g. When battery is empty and solar production is zero).

So I think, changing the automation "input_number.set_sg_forced_charge_discharge_power" to float only corrects to automation. It does not affect the forced charging. Forced charging works independent from the float bugfix, one just have to wait some more minutes!

hAMMERlized, 2024-01-03:
lol that's what I'm telling you since the last posts. It takes a few minutes until it charges the battery...

bliblabl8, 2024-01-03:
Sry, my bad. Imho you mentioned 10 seconds, not 10 minutes. Anyways, forced charging is working for me. Thank you for this great Integration!

Crootie, 2024-05-27:
Just did the latest update for home assistant:
[image]
 Tried to force  charge and lost it again. Was working great and left it open in case others had issues with the same thing.
[image]
[image]

lassej01, 2024-05-27:
Change this one, and then you go :-)
[image]

Crootie, 2024-05-27:
Just ran that to 5000 and it is working....... but... that's discharging not charging. Quick education please.

Crootie, 2024-05-27:
actually don't bother.. it is a dual setting depending on charge or discharge... thankyou.

lassej01, 2024-05-27:
[image]
Maybe it should be an "&" in the middle for clearifaction.

Crootie, 2024-05-27:
even just a charge/discharge would work it. but thankyou for your assistance