mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #167: 2023.10

Mats1717, 2023-09-29:
Seems like it is a modbus errer again when updating to 2023.10b02
Had to roll back to have it working again.

mkaiser, 2023-09-29:
Can you provide us an error description or log?

Mats1717, 2023-09-29:
Sorry but I went back to last days backup so i dont have it. Quit sure if you install 2023.10.0b2 you will have the same problem too.

Mats1717, 2023-09-29:
Installed it again

Logger: homeassistant.setup
Source: setup.py:185
First occurred: 19:35:56 (1 occurrences)
Last logged: 19:35:56

Logger: homeassistant.config
Source: config.py:501
First occurred: 19:35:56 (1 occurrences)
Last logged: 19:35:56

Invalid config for [modbus]: BDC rated power: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][19]. Got {'name': 'BDC rated power', 'unique_id': 'sg_bdc_rated_power', 'slave': 1, 'address': 5627, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 100, 'scan_interval': 600} BMS max. charging current: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][20]. Got {'name': 'BMS max. charging current', 'unique_id': 'sg_bms_max_charging_current', 'slave': 1, 'address': 5634, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60} BMS max. discharging current: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][21]. Got {'name': 'BMS max. discharging current', 'unique_id': 'sg_bms_max_discharging_current', 'slave': 1, 'address': 5635, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60} Backup phase A power: `count: 1` cannot be combined with `data_type: int16` @ data['modbus'][0]['sensors'][23]. Got {'name': 'Backup phase A power', 'unique_id': 'sg_backup_phase_a_power', 'slave': 1, 'address': 5722, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10} Backup phase B power: `count: 1` cannot be combined with `data_type: int16` @ data['modbus'][0]['sensors'][24]. Got {'name': 'Backup phase B power', 'unique_id': 'sg_backup_phase_b_power', 'slave': 1, 'address': 5723, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10} Backup phase C power: `count: 1` cannot be combined with `data_type: int16` @ data['modbus'][0]['sensors'][25]. Got {'name': 'Backup phase C power', 'unique_id': 'sg_backup_phase_c_power', 'slave': 1, 'address': 5724, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10} Battery capacity: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][112]. Got {'name': 'Battery capacity', 'unique_id': 'sg_battery_capacity', 'slave': 1, 'address': 33048, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'unit_of_measurement': 'kWh', 'device_class': 'energy', 'scale': 0.01, 'scan_interval': 600} Battery charging start power: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][113]. Got {'name': 'Battery charging start power', 'unique_id': 'sg_battery_charging_start_power', 'slave': 1, 'address': 33148, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10} Battery current: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][84]. Got {'name': 'Battery current', 'unique_id': 'sg_battery_current', 'slave': 1, 'address': 13020, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 1, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 0.1, 'scan_interval': 10} Battery discharging start power: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][114]. Got {'name': 'Battery discharging start power', 'unique_id': 'sg_battery_discharging_start_power', 'slave': 1, 'address': 33149, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 10, 'scan_interval': 10} Battery forced charge discharge cmd raw: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][103]. Got {'name': 'Battery forced charge discharge cmd raw', 'unique_id': 'sg_battery_forced_charge_discharge_cmd_raw', 'slave': 1, 'address': 13050, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'scan_interval': 10} Battery forced charge discharge power: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][104]. Got {'name': 'Battery forced charge discharge power', 'unique_id': 'sg_battery_forced_charge_discharge_power', 'slave': 1, 'address': 13051, 'input_type': 'holding', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scan_interval': 10} Battery level: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][86]. Got {'name': 'Battery level', 'unique_id': 'sg_battery_level', 'slave': 1, 'addre [... cut ...] be combined with `data_type: uint32` @ data['modbus'][0]['sensors'][48]. Got {'name': 'Yearly PV generation (2029)', 'unique_id': 'sg_yearly_pv_generation_2029', 'slave': 1, 'address': 6277, 'input_type': 'input', 'data_type': 'uint32', 'swap': 'word', 'count': 2, 'precision': 1, 'unit_of_measurement': 'kWh', 'device_class': 'energy', 'state_class': 'total_increasing', 'scale': 0.1, 'scan_interval': 600}. (See ?, line ?). Please check the docs at https://www.home-assistant.io/integrations/modbus

mkaiser, 2023-09-29:
Thank you very much for reporting and the error log!

similar to the last modbus changes, the new changes are all very reasonable. The modbus integration now seems much better and has reached a 100%  code coverage.

The latest changes in the upcomping HA 23.10 will add more yaml checks. 
In the current core of this integration, there are some irregularities, that are now (correctly!) marked as errors. 

Will fix them over the weekend, so hopefully a usable version will be out when the 23.10 HA version is officially released

Mats1717, 2023-09-29:
👍

mkaiser, 2023-09-29:
... that was way easier than anticipated from the huge log :)

I basically just removed around 200 lines of code:

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/commit/53522010455c80e1575347ac5355ead854db95df

tested with
Home Assistant 2023.10.0b2
Supervisor 2023.09.3
Operating System 10.5

can you double check?

Mats1717, 2023-09-30:
Tested now and I can confirm its working :)

closed 2023-09-30 by mkaiser