mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #169: state_class not defined for some sensors

phil11c, 2023-09-30:
A lot of sensors do not have state_class (measurement, total or total_increasing) defined.
For that reason no long-term statistics are available for those sensors.

Not sure if this is intentionally - if so, i will keep on adding this after every new release.

mkaiser, 2023-09-30:
which ones are you missing?

phil11c, 2023-10-01:
Thanks for your fast reply.

sg_inverter_temperature
sg_mppt1_voltage
sg_mppt1_current
sg_mppt2_voltage
sg_mppt2_current
sg_bms_max_charging_current
sg_bms_max_discharging_current
sg_battery_voltage
sg_battery_current
sg_battery_temperature
sg_export_power_limit_mode_raw
sg_battery_capacity
sg_mppt1_power
sg_mppt2_power
sg_phase_a_power
sg_phase_b_power
sg_phase_c_power
sg_battery_charge_nom
sg_battery_charge

mkaiser, 2023-10-02:
I guess it won't hurt to add the state_classes there.

but 
sg_bms_max_charging_current
sg_bms_max_discharging_current

never changed in my system. Are they changing in yours?

phil11c, 2023-10-02:
Thanks for adding them.

A battery is on my roadmap, right now i don't have one.
So no, they are not changing :-)

Louisbertelsmann, 2023-10-02:
They're changing, at least on the Sungrow battery, when the bms limits charging and discharging speeds, for example when it's too cold for the battery, the BMS limits the charging current.

closed 2023-11-04 by mkaiser