mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #170: Sensors yearly

Mats1717, 2023-10-01:
Hi
It seems like these sensors are way off, shows more than double value.
sensor.yearly_export_current
sensor. yearly_export_2023
sensor.yearly_pv_generation_current
sensor. yearly_pv_generation_2023
For me total exported shows 5635,2 and yearly shows 14980,4
Total PV generation 7789,4 and yearly shows 17133,1
Plant started in February 2023
Is it because it is first year and some values where in the inverter at startup.

C-l-a-u-s, 2023-10-04:
I have the same error.
PV export is much too high. I have seen that the 1st day of each month with extreme values drive up the total value...

mkaiser, 2023-10-04:
Please use the template for bug reports like this

Without more information (firmware, version etc....) , I cannot do anything to identify the issues :/

C-l-a-u-s, 2023-10-19:
> Please use the template for bug reports like this (create a new issue or edit your post to add the information)
> 
> Without more information (firmware, version etc....) , I cannot do anything to identify the issues :/

I have now done a firmware update of the inverter and now it works again. Thanks a lot!

Mats1717, 2023-10-20:
> Hi It seems like these sensors are way off, shows more than double value. sensor.yearly_export_current sensor. yearly_export_2023 sensor.yearly_pv_generation_current sensor. [...]

Now my supplier have done a firmware update of the inverter and now it works. :)

closed 2023-10-20 by Mats1717