mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #172: Wrong rounding in Battery Charge Sensor

drschlaumeier, 2023-10-02:
Hi, I discovered that you have a small error in rounding the Battery Charge.
It currently rounds the (max_soc - min_soc)/10000 instead of (battery_capacity * (max_soc - min_soc)/10000).
so, if you have 100%-5% it rounds to 0.9 which gives wrong capacity...

BR

```

      - name: "Battery charge"
        unique_id: sg_battery_charge
        unit_of_measurement: kWh
        device_class: energy
        availability: >-
          {{
              states('sensor.battery_capacity') | is_number and
              states('sensor.battery_level') | is_number and
              states('sensor.min_soc') | is_number and
              states('sensor.max_soc') | is_number
          }}
        state: >-
          {{  
              states('sensor.battery_capacity') | float *
                ((states('sensor.max_soc') | float - states('sensor.min_soc') | float) * 
                states('sensor.battery_level') | float / 10000) | round(1)
          }}
```

mkaiser, 2023-10-02:
you are right, thanks for the hint!

just to be sure, because I am a bit unconcentrated at the moment... So it should look like this?

(see last section, I used the first lines for the developer tools --> template window)

sometimes the syntax is quiet challenging and hard to read :/

> 
> "{{ states('sensor.battery_capacity') }}"
> "{{ states('sensor.max_soc') }}"
> "{{ states('sensor.min_soc') }}"
> "{{ states('sensor.battery_level') }}"
> 
> {{ "(max-min) " }} 
> "{{ states('sensor.max_soc') |float - states('sensor.min_soc') |float  }}"
> 
> {{ "(max-min)    * capacity" }} 
> "{{ states('sensor.battery_capacity') |float *
>   ( states('sensor.max_soc') |float - states('sensor.min_soc') |float ) /100
> }}"
> 
> 
> {{ "(max-min)   * capacity   * batt level" }} 
> "{{ states('sensor.battery_capacity') |float *
>   ( states('sensor.max_soc') |float - states('sensor.min_soc') |float ) /100 *
>    states('sensor.battery_level') |float /100
> }}"
> 
> 
> {{ "((max-min)   * capacity   * batt level   ) | round" }} 
> "{{ ( states('sensor.battery_capacity') |float *
>   ( states('sensor.max_soc') |float - states('sensor.min_soc') |float ) /100 *
>    states('sensor.battery_level') |float /100 
>    ) | round(2)
> }}"

drschlaumeier, 2023-10-04:
Sorry for late reply. Best is you set brackets in each step calculation since the HASS template seems be stupid :-)

e.g. to be sure that the final result will be rounded instead the max-min ...

{{ "((max-min) * capacity * batt level ) | round" }}

"{{ 
     ( 
          (states('sensor.battery_capacity') |float) *
        ( 
               ((states('sensor.max_soc') |float) - (states('sensor.min_soc') |float ) /100) *
               ((states('sensor.battery_level') |float) /100)
        )
     ) | round(2)
}}"

closed 2023-11-04 by mkaiser