mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #173: Invalid config for [modbus]: BDC rated power

MrMontesa, 2023-10-05:
**Describe the bug**
Hey, I did an update this morning to my homeassistant install (2023.10.0) as well as upgradeed the firmware of my BYD battery and powercycled the whole PV system.
After updating battery firmware, I think all was working fine. But after the HASS update I got the below error message:
```
Invalid config for [modbus]: BDC rated power: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][19]. Got {'name': 'BDC rated power', 'unique_id': 'sg_bdc_rated_power', 'slave': 1, 'address': 5627, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 100, 'scan_interval': 600} BMS max. charging current: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][20]. Got {'name': 'BMS max. charging current', 'unique_id': 'sg_bms_max_charging_current', 'slave': 1, 'address': 5634, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60} BMS max. discharging current: `count: 1` cannot be combined with `data_type: uint16` @ data['modbus'][0]['sensors'][21]. Got {'name': 'BMS max. discharging current', 'unique_id': 'sg_bms_max_discharging_current', 'slave': 1, 'address': 5635, 'input_type': 'input', 'count': 1, 'data_type': 'uint16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'A', 'device_class': 'Current', 'scale': 1, 'scan_interval': 60} Backup phase A power: `count: 1` cannot be combined with `data_type: int16` @ data['modbus'][0]['sensors'][23]. Got {'name': 'Backup phase A power', 'unique_id': 'sg_backup_phase_a_power', 'slave': 1, 'address': 5722, 'input_type': 'input', 'count': 1, 'data_type': 'int16', 'swap': 'none', 'precision': 0, 'unit_of_measurement': 'W', 'device_class': 'power', 'state_class': 'measurement', 'scale': 1, 'scan_interval': 10} Backup phase B power: `count: 1` cannot be combined with `data_type: int16` @ data['modbus'][0]['sensors'][24]......
```

**Your Sungrow inverter:**
 - Model: SH-10.RT
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [] WiNet-S (LAN)
   - [X] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [X] yes
  - [] no 
  - [] I don't know what that is

**Home Assistant :**
 - Version: 2023.10.0

**modbus_sungrow.yaml:**
 - Version/ time stamp : by Martin Kaiser, # last update: 2023-09-12

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Not sure, I just did the Home Assistant Update

Do you have an idea what could be wrong here? Currently all my sensors show "unavailable.
Thanks for your help

MrMontesa, 2023-10-05:
seems like I found the related core issue: https://github.com/home-assistant/core/issues/101421

Louisbertelsmann, 2023-10-05:
Update to the newest version of the modbus_sungrow.yaml

MrMontesa, 2023-10-05:
Thanks for your swift reply. After upgrading to the most recent version, everything is back.

closed 2023-10-05 by MrMontesa