mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #176: Forced charging not charging at full power

Tsopic, 2023-10-06:
### Discussed in https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/175

<div type='discussions-op-text'>

<sup>Originally posted by **Tsopic** October  6, 2023</sup>
Not sure what's wrong with this setup. 

Maybe someone has ideas.
[image]

[image]

[image]
[image]

[image]

It should be currently importing 10000w but it does only 1k.

I am looking forward to any ideas for a solution, if anyone has some similar issues with forced charge.
</div>

mkaiser, 2023-10-06:
The settings look good to me. 

What are your BMS current values?

[image]

Please edit your first post and let us know the values from the issue template. You did not share much about your setup:  inverter type, battery type,  LAN cable attachment, ...

What happens, if you are using the sungrow app and not this integration?

Are you sure you have the newest firmware for both, inverter and battery?

Once in my setup, the battery and inverter were kind of out-of-sync. A cold reboot (including power off for some minutes) helped.  Did you try this?

Tsopic, 2023-10-07:
I have not tried the cold reboot yet, updated the original post with Battery Amps and setup details. If need more info, I'm happy to provide it.

It's a strange issue since 1 day before in the morning it was chugging energy from the grid with full power.  And the day before data also with only 50%.

[image]

Tsopic, 2023-10-08:
Adding some more info from isolarCloud. 

After hard resetting it - powering off for 5min and powering on + setting discharge limiter lower and then back to 10k. It does discharge with full power again.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Now it refuses to consume from battery for some reason 😮‍💨 

Lemme know if there are some protection parameters I should change or improve.

closed 2024-12-15 by github-actions[bot]