mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #182: I don't get it: Battery level (nominal)

elektrinis, 2023-10-13:
**Describe the bug**
Just loaded the latest code.
Battery is 100% charged (real 100% when soc limit was set to 100%)

SH10RT connected via internal LAN.
Latest firmware
Latest HAOS

extra issue: inverter temperature is Unknown (same as before)

[image]

Louisbertelsmann, 2023-10-13:
The Battery level is the real usable battery percentage, if you set min SoC to 10% and max SoC to 90%, 80% of the battery are usable. If the battery is 50% charged, that then makes the Battery level (nominal) 40%. 
About the temperature being unavailable I don't know anything.

mkaiser, 2023-10-14:
[elektrinis](https://github.com/elektrinis), can you join the discord channel (link at readme page) to discuss the temperature issue? As far as I remember your issue is very old and I guess we can better discuss this in a chat

elektrinis, 2023-10-15:
I'll join when I have some spare time, thanks for inviting.

About the SOC. I'm perhaps misunderstanding the idea, but... My battery was set to 0/100% limits and charged fully. After I set upper limit to 90%, and if battery is still charged to real 100%, I would expect this 'normalized' value to show 111%, and not 90%. It seems to be in reverse?

Louisbertelsmann, 2023-10-15:
The inverter unfortunately doesn't publish the real SoC, just the relative one.

mkaiser, 2023-10-16:
elektrinis, can you poste a screenshot of these two entities in comparison? In my case I have set min Soc to 9% and max Soc to 92 %. So these values are kind of reasonable in my case. 
[image]

just found this in a SH8RS datasheet. https://www.rpc.com.au/pdf/sungrow_SH8-10RS_manual.pdf 
[image]

Should we rename it like this?
battery level --> usable Soc
battery level (nominal) --> real Soc

elektrinis, 2023-10-16:
[image]

From your screenshot, I would say that `battery level nominal` should be `battery real SOC`, and `battery level` should be `battery usable SOC`. Usable energy is always lower than max chemical energy, thus percentage should always be higher.
The word "nominal" is bit misleading, I expect that nominal is based on actual capability of the battery, disregarding any limits. But I'm not sure where are these numbers coming from now, perhaps it's the inverter where these are born?

To my taste, `battery level` should always show SOC of actual hardware. For example, if `min soc` = 0% and `max soc = 90%`, when charged to this limit, I would expect `battery level` to be 90% and `battery level (usable)` to be 100%, because it is at 100% compared to the limits.

In my case, both min/max SOC were set to 0/100%. When battery was fully charged to 100%, both SOCs showed 100%, which is OK. But when `max soc` was lowered to 90%, I was expected that `battery level` would remain at 100%, and `battery level (usable)` would jump to 111%, as the charge was actually higher than set by the `max soc`.

Also not sure what to do with `reserved soc`. I think it should be included in the calculation together with min/max SOC. But I don't use this relative SOC in my automations, so not sure what others think.

[image]
Now I had `min SOC` = 0 and `max SOC` = 90%, it showed 100/90%.
Then I have set `max SOC` = 100%, and suddenly both values jumped to 100%, and then both to 90%. Not sure what is happening here, will have to investigate more.

Disclaimer: I design battery management systems for a living.

elektrinis, 2023-10-16:
Did a quick test now when actual real battery SOC was 92%. Lowered `max SOC` to 80% and got this weird behavior. `nominal` jumped DOWN (was expecting one of them to jump up), and a minute later it started showing 100/80% - both being wrong. Once I set `min SOC` back to 100%, both changed to 92%. I'm lost here, developing trust issues now.

[image]
[image]

elektrinis, 2023-10-16:
OK guys, after staring at it for some time, few hours later I think I'm starting to get it.
I'm missing some information, but from what I see it looks like inverter is only returning relative SOC and this integration is trying to calculate the real ("nominal") SOC from the information it gets.

I was confused with the "nominal" term and the fact that inverter caps it at 100% and behavious becomes weird when min/max SOC is adjusted has fueled the confusion further. But now this makes sense.
I'll just leave it at 0/100% and do the limiting in my automations as needed.
Sorry for wasting your time.

amo13, 2023-12-07:
Unfortunately, I still don't get it...

Is the "nominal" value an estimated calculation made by this integration and nothing with that name is actually coming as data from the inverter?

How can we confirm that setting min and max SoC actually works?

And what is the difference between min SoC and reserved SoC? If I set min SoC to 5%, reserved SoC to 20% and max SoC to 10%, do I have 65% usable capacity? What exactly would a SoC of 0%, 20% or 100% under these circumstances mean?

closed 2024-12-08 by github-actions[bot]