mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #183: After upgrade to OS 11 and HA 2023.10.3 no data

hboltz, 2023-10-15:
**Describe the bug**

After upgrade to OS 11 and HA 2023.10.3 no data

**Your Sungrow inverter:**
 - Model: [SH-8.0RT v112]
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [X] I don't know what that is

**Home Assistant :**
 - Version: [2023.10.3]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-29] (see header of the file)

** Inverter Firmware Status:**
   - [X] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Louisbertelsmann, 2023-10-15:
With this "description", or rather the lack of it, it could be basically everything. Would you please describe it more precisely?

hboltz, 2023-10-15:
After Reset ok

closed 2023-10-15 by hboltz