mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #184: Support for SH10RS

ABredhauer, 2023-10-16:
I've just had a SH10RS and battery installed and am finding there's a number of quirks that look like this particular (very new) inverter isn't fully supported by the current integrtion. 

First of which is the inverter has 4MPTTs and I can only see 2 reported by the integration

I'm also seeing some strange values reported such as 
[image]

I'm also getting a large number of Index out of range and Unable to decode request errors in my logs

I realise there's not many of these inverters around yet so happy to help work through adding the support but also trying to get my head around how this integration works at the same time. 

I'm also running into issues where the LAN connection from the inverter simply drops off my network randomly. So far my only solution is to unplug the dongle or shutdown and restart the inverter

frankyman88, 2023-11-02:
Getting mine installed in two weeks so keen to see how this goes!!

bleughb, 2023-11-07:
The integration is surprisingly easy. the code modbus_sungrow.yaml holds 90% of the stuff

youread the inverter modbus register manual  then add a line in the yaml file to cover what you want

For example, in this code -Voltage
MPPT1 seems to be 5011
MPPT2 is 5012

A quick google shows these following registers at another github
so, copy / paste the existing ones, alter the number, reboot home assistant and see what happens!
you'll need to ad dthe new sensors in the dashboard

ModBusRegister(5011, "mppt_1_voltage", "U16", 0.1, VOLTAGE),
ModBusRegister(5012, "mppt_1_current", "U16", 0.1, AMPERE),
    ModBusRegister(5013, "mppt_2_voltage", "U16", 0.1, VOLTAGE),
    ModBusRegister(5014, "mppt_2_current", "U16", 0.1, AMPERE),
    ModBusRegister(5015, "mppt_3_voltage", "U16", 0.1, VOLTAGE),
    ModBusRegister(5016, "mppt_3_current", "U16", 0.1, AMPERE),
    MODBusRegister(5115, "mppt_4_voltage", "U16", 0.1, VOLTAGE),
    ModBusRegister(5116, "mppt_4_current", "U16", 0.1, AMPERE),

ABredhauer, 2023-11-07:
Doesn't seem to be as straight forward as this. For some reason my inverter is returning illegal register codes for the MPPT 3 and 4 registers you've listed, not sure if another firmware update is required or not

> I'm also running into issues where the LAN connection from the inverter simply drops off my network randomly. So far my only solution is to unplug the dongle or shutdown and restart the inverter
I was able to fix/reduce these issues by commenting out the backup registers which were also being reported as illegal. Seems like too many errors was causing an issue for me

bleughb, 2023-11-07:
> Doesn't seem to be as straight forward as this. For some reason my inverter is returning illegal register codes for the MPPT 3 and 4 registers you've listed, not sure if another firmware update is [...]

ok. I'd have a good google about then as the registers seem to change a little depending on the inverter.....
RE your LAN connection shutting down 
- delete all non working / required registers
- change the polling time to 60s for most registers
- change the polling time to 600s or more for the registers that don't need interrogating as much
I've found the winet-s connection seems to lock up if you interrogate it too much

ABredhauer, 2023-11-07:
Yes commenting out the 4 sensors relating to backup power seems to have made a big difference. I am still getting some modbus errors in my logs so there are obviously still some sensors that are not supported but I haven't been able to find them yet. 

My inverter got a firmware update overnight but it doesn't seem to have made a difference as yet. I'm guessing it is the WINet module which needs the firmware update to resolve some of these issues now but it doesn't have one yet.

bleughb, 2023-11-08:
Maybe try downloading a modbus scanner and start sniffing registers?
one just helped me with a weird register 5113 / running time issue on my SG5.0RS devices....
https://www.se.com/au/en/faqs/FA180037/ has a great one (look at the end)

ABredhauer, 2023-11-09:
> Maybe try downloading a modbus scanner and start sniffing registers?

This is what I'm doing. So far I'm finding what is in the doc doesn't match how the inverter is behaving. Not sure if I can ask Sungrow for help here or if they will tell me to go away

mkaiser, 2023-11-09:
It won't hurt asking some sungrow guys to send you an updated modbus register description - they are kind of secretive with these information, so I don't have a recent version, too...

ABredhauer, 2023-11-09:
Sorry I found a more recent version and downloaded it but need to find the link for you

ABredhauer, 2023-11-10:
Found it, attached here and the original link I found it from was https://www.photovoltaikforum.com/thread/166134-data-read-from-sungrow-inverters-modbus/? postID=3452929#post3452929 
[TI_20231019_Communication.Protocol.of.Residential.Hybrid.Inverter_V1.1.2_EN.pdf](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/13314565/TI_20231019_Communication.Protocol.of.Residential.Hybrid.Inverter_V1.1.2_EN.pdf)

ABredhauer, 2023-11-12:
For anyone else with this model inverter I've found the cause of the remainder of the modbus errors I was getting. In the latest doc (above) there are a bunch of registers that are listed as not supported when using Ethernet via the WINET-S which is unfortunately the only option we have. So I've now had to comment out

- total backup power + phase A,B,C Backup power
- Monthly PV generator for Jan - Dec
- Yearly PV Generation (2019 - 2029)
- Monthly Export for Jan - Dec
- Yearly export (2019 - 2028)

I've also had to change the battery capacity sensor to an input sensor on register 13038

frankyman88, 2023-11-23:
Hey Team,
Finally got my system up and running!
Wanted to be able to delay the battery charging start time to later in the day so I can export more power in the morning and charge the battery off the clipped output (I have 18kW of panels and inverter is only 10kw).
Sungrow's local webapp and isolar cloud app are infuriating to say the least!
I have been configuring this addon steadily following the advice here and removing lines that don't seem to work.
**However, I need the EMS mode select and it was throwing a KeyError: 'SungrowSHx'
I was able to fix it by changing "hub: SungrowSHx" to "hub: SH10RS"**
Should apply to all the automations, such as change force discharge command, all the sliders etc.
Now I can just schedule force stop battery from 7am until 10am and then set it back to self consumption for charging with an automation :) 
Thanks for your hard work all!

dylan09, 2023-11-23:
I realized the delayed charging using sg_battery_charging_start_power and sg_battery_max_charge_power.

The first one is for efficiency always set to 200 W. For me it makes no sense to let the inverter start charging below 200W (arbitrarily defined by me).

If I would like to delay charging I set sg_battery_max_charge_power to 100 W. And when charging should start at noon I set it back to the power I like to charge with.

ABredhauer, 2023-11-25:
@dylan09, Do you have a SH10RS that you were able to get to follow the max battery charge power setting? I have tried setting mine as low as 100W and it just ignores the setting and sends everything to the battery.

I have a case with Sungrow about this but for now I am using the same approach as @frankyman88 and setting the EMS mode to forced early in the morning and then setting it back to self consumption when I want to start charging the battery

dylan09, 2023-11-25:
Sorry. I am on the wrong thread. Missed that you have SH10RS.

reesaroo74, 2023-11-26:
> @dylan09, Do you have a SH10RS that you were able to get to follow the max battery charge power setting? I have tried setting mine as low as 100W and it just ignores the setting and sends everything [...]

> For anyone else with this model inverter I've found the cause of the remainder of the modbus errors I was getting. In the latest doc (above) there are a bunch of registers that are listed as not [...]

How did you get on with things like Grid Frequency and Battery Discharging Start Power?

I'll have a crack at your fix for the battery capacity sometime this week.  Thanks for the tip!

[image]

[image]

dylan09, 2023-11-27:
> @dylan09, Do you have a SH10RS that you were able to get to follow the max battery charge power setting? I have tried setting mine as low as 100W and it just ignores the setting and sends everything to the battery.
> 
The HA automation is triggered when you change max battery charge? There should be a message in the log file.

Did you check if you see the new value in iSolarCloud?

ABredhauer, 2023-11-28:
> How did you get on with things like Grid Frequency and Battery Discharging Start Power?
> 
I found some other comments on here for another issue that the Grid frequency is out by a factor of 10 when using the WInet dongle which is our only option so I had to change the scale for that sensor
scale: 0.01

Battery charge/discharge start power are not anywhere I can see in the latest modbus docs so I have not done anything with those yet. I've checked the registers and they're not valid on my inverter so I have no idea how to get that value

> I'll have a crack at your fix for the battery capacity sometime this week. Thanks for the tip!
> 
> I'm also having trouble with my Daily's.
> 

In terms of the daily import/export figures I mainly got HA to calculate those so hadn't noticed the registers are returning 0. I've just checked now and the registers are correct according to the doc but both are zero on mine which is not correct so I have the same behaviour as you. I'll add these to my chat with Sungrow

frankyman88, 2023-11-28:
Has anyone been able to get total DC power working?
Home assistant shows "entity not available: sensor.total_dc_power"
However, I am able to poll the register 5017 and get an accurate result with the tester linked higher up in this thread by bleughb. I have been trying various combinations of byte/word swapping with data_type uint32, and also some custom ones without any luck!

As with Abredhauer, I can't get any of the max charging or start power options working either.
Register 33047 returns a value of 106 which I assume corresponds to charging rate of 10.6kW
However, register 33048 returns value of 1. I changed it to 5 and it didn't do anything... still discharging at full speed. Maybe it is overridden by the "whole home backup" setting.
I will try reducing the charge rate in the morning from 106 to 1 which should limit the charge to 100W (fingers crossed), then can use @dylan09's solution which would be better in a blackout situation as it would still allow discharging instead of the full forced stop.

@reesaroo74 I have put the code below for working battery capacity and grid frequency entities :) 
      _- name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 13038 # reg 33049
        input_type: input
        data_type: uint16
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.01
        precision: 0
        scan_interval: 600_

**Grid frequency:** 
      _- name: Grid frequency
        unique_id: sg_grid_frequency
        device_address: !secret sungrow_modbus_slave
        address: 5035 # reg 5036
        input_type: input
        data_type: uint16
        precision: 2
        unit_of_measurement: "Hz"
        device_class: frequency
        state_class: measurement
        scale: 0.01
        scan_interval: 10_

frankyman88, 2023-11-28:
Bit of an update on charging speed, I was out by a factor of 10 (10.6kW equates to value 1060).
Regardless the settings appear to do nothing. 
I even found it in ISolarCloud settings, changed it there and still no effect. 
So I'm guessing it is a fault with the new SH10RS inverter and we need to wait for firmware update to fix it

ABredhauer, 2023-11-28:
> So I'm guessing it is a fault with the new SH10RS inverter and we need to wait for firmware update to fix it

well I’m glad to see you’ve got the same result as me but at the same time was hoping it would work and I had missed something. Would be good if you could also raise it with Sungrow so they know there’s a few of us trying to get it working

frankyman88, 2023-11-29:
Submitted a request for more modbus info or another way to control battery charge time to utilise excess DC power which would be wasted. I also asked them how long it will before they partner with Ergon Energy here in QLD to provide dynamic export support (seems to be exactly the same as South Australia which is already active!)

Johnnyd1965, 2023-12-31:
Has anyone managed to get a copy of the Modbus info for the SH10RS?

ABredhauer, 2023-12-31:
> Has anyone managed to get a copy of the Modbus info for the SH10RS?

The latest I have is above https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/184#issuecomment-1804969769

Is there something specific you're after? My attempts to get anything further out of Sungrow still haven't gotten anywhere

frankyman88, 2023-12-31:
@ABredhauer I was fiddling today with the charging rates again. When the EMS is in self consumption mode you can't control anything, but if you set forced mode and forced charge setting, register 13052 (Charge/discharge
power) then manages the charging speed! It does some funny stuff at the lower end (100 =1000W), but past about 5000 it is 1:1 scale. I'm going to try setting it to 2000 at 10am, 3000 at 11am etc and then scaling down from 1pm onward. Happy new year!

frankyman88, 2024-01-01:
I put my graduated charging speed settings (changes based on time and whether or not it is sunny) into a yaml  [here https://github.com/frankyman88/SungrowMisc/tree/main](url). I am on an Ergon time of use tariff (12C solar soaker) which makes power cheap between 9-4pm, so I have it set to force charge max speed from 2-4pm to top up from grid if needed. 
I also wanted to keep some battery power in reserve but not at the expense of buying grid energy before 9pm, so reserved SOC is 10% before 9pm and switches to 35% after, conserving whatever I have left. Hope you might get some inspiration at least!

jbekkema, 2024-01-01:
> When the EMS is in self consumption mode you can't control anything, but if you set forced mode and forced charge setting, register 13052 (Charge/discharge power) then manages the charging speed!

Do you by chance know the behaviour of the inverter in the following two scenarios when using that approach?

1. The battery is in forced charging mode with a limit in place, and a mains blackout occurs. Will it still switch to backup mode and start discharging the battery at a rate that matches the house load?

2. The battery is in forced charging mode with a limit in place, with the solar output covering house load and exporting to grid. Sudden cloud cover drops the output of the solar, causing the output to be less than the house load. Will it start drawing what it needs from the battery, or will it instead draw it from the grid?

My current approach is to set the Battery Max SOC value to 50% (as low at it goes), and then ramp it up to 100% around midday. Definitely not an ideal approach, but it doesn't appear to cause any issues with backup operation when cutting off mains or cutting off the solar panels. It would definitely be nicer to be able to put a limit on the charge rate.

ABredhauer, 2024-01-01:
Thanks @frankyman88 this looks great. My only concern with this approach is the possibility of drawing from the grid to charge the battery - sounds like its not really an issue for you though. I have been thinking about an alternative using the Max SOC setting to gradually ramp up production during the day but the weather here hasn't been great so I haven't had a chance to test it yet

> When the EMS is in self consumption mode you can't control anything,
Is this official from Sungrow or what you have noticed? I was wondering if this was the case but was hoping it wasn't. Sungrow support haven't been super helpful on why the max charging is being ignored but from their responses they seem to expect it to be honoured by the inverter so I'm hopeful of a fix coming eventually

@jbekkema I haven't tested your scenarios but I suspect you would need to set the inverter back to self consumption for it to draw from the battery. The way I understand it is that in the forced modes the battery will only do what you tell it to. That said I've seen twice now that when in forced discharge mode my inverter will sometimes send some power to the battery

Johnnyd1965, 2024-01-02:
I was after the modbus addresses for MPPT3 and MPPT4.  Has anyone worked this out?

jbekkema, 2024-01-02:
> I was after the modbus addresses for MPPT3 and MPPT4. Has anyone worked this out?

The addresses for MPPT 3 listed previously above are correct. However when using Modbus over TCP a valid value is not returned. If using Modbus over a web socket it works correctly. Coding up a Modbus proxy that relays TCP commands to the web socket might be a workaround, but probably not worth the effort for just the MPPT 3 values.

The addresses for MPPT 4 listed previously don't appear to be correct for the SH10RS: neither Modbus over TCP, nor over the web socket. They're also not listed in the "Communication Protocol of Residential Hybrid Inverter" document linked above (however they are listed in the docs for the non-hybrid models). It's possible the SH10RS may list the values for MPPT 4 on a different register number, or not at all. If scanning to see if you can find the correct address, you'll likely need to use the web socket (or via the Winet web UI under Device Monitoring->SH10RS->General Parameters).

frankyman88, 2024-01-02:
> Do you by chance know the behaviour of the inverter in the following two scenarios when using that approach?
> 
>   The battery is in forced charging mode with a limit in place, and a mains blackout occurs. Will it still switch to backup mode and start discharging the battery at a rate that matches the house load?

I just simulated a blackout by switching off my grid supply, while my house load was higher than total DC power. The inverter state changed from "Forced Charge" to "Off Grid" instantly, and started discharging the battery to make up the difference in power. No devices in my house experienced an impact, and my UPS didn't trigger etc. So that is a relief!

> The battery is in forced charging mode with a limit in place, with the solar output covering house load and exporting to grid. Sudden cloud cover drops the output of the solar, causing the output to [...]

I haven't been able to test this conclusively, but I from looking at my data it would choose to pull from grid instead of battery if you have mode set to "Forced Charge" or "Stop"
> My current approach is to set the Battery Max SOC value to 50% (as low at it goes), and then ramp it up to 100% around midday. Definitely not an ideal approach, but it doesn't appear to cause any [...]

Yeah that's fair enough, I'm just trying to restrict the rate enough to get my battery full without causing too much drop off in Exports. I have a few plant sensors which report illuminance, might give that a go with my automation instead of the weather integration -> stop altogether if <1000lux, charge at 1000w if between 1000-2000lux, 3000w between 2000-3000lux and no restriction above 4000lux. 

> Is this official from Sungrow or what you have noticed?

Just what I've noticed, I agree I hope that they fix it too as their support seemed to imply that those max charging power registers should work *shrug*

jbekkema, 2024-01-02:
> The addresses for MPPT 4 listed previously don't appear to be correct for the SH10RS

Following up on my own post, I did some scanning via the web socket, and the following registers (all UInt16, 0.1 Scale) appear to hold the correct MPPT values:

MPPT 1 Voltage Register: 5333
MPPT 1 Current Register: 5334
MPPT 2 Voltage Register: 5335
MPPT 2 Current Register: 5336
MPPT 3 Voltage Register: 5337
MPPT 3 Current Register: 5338
MPPT 4 Voltage Register: 5339
MPPT 4 Current Register: 5340

However frustrating they only appear accessible via the web socket, not via Ethernet (I have not tested via Wi-Fi).

jbekkema, 2024-01-02:
> I just simulated a blackout by switching off my grid supply, while my house load was higher than total DC power. The inverter state changed from "Forced Charge" to "Off Grid" instantly, and started [...]

That's good to know, thanks for testing!

> I haven't been able to test this conclusively, but I from looking at my data it would choose to pull from grid instead of battery if you have mode set to "Forced Charge" or "Stop"

Ah, damn. You could test it by flipping the DC isolator on the left to Off to cut the power from your panels, but it sounds like if you've seen it drawing from the grid instead of the battery at times of low solar output (or high house usage) then it's likely unnecessary.

Hopefully Sungrow can fix it in a future firmware update so the standard charge/discharge values are not ignored, as that would be the ideal solution.

Johnnyd1965, 2024-01-03:
My work around is to use gosungrow for getting ant missing info.

NickStallman, 2024-03-26:
I just got my SH10RS and yeah the inconsistent sensors are quite annoying.

Does this affect the inverter as a whole? Or is it purely from the WiNet-S2?
I'm wondering if it's worth tapping in to the RS485 port (A1 and B1) if that'll give all the correct values, and then I can just use Esphome to feed them over to Home Assistant.

SNAPE15, 2024-04-05:
has anyone got Daily Imported Energy working? I have figures for Total Imported Energy but nothing showing for daily. Doesn't even show via iSolarCloud website (yet total does).

danielclau, 2024-04-05:
I use Total Imported Energy in the Energy Dashboard.
This in turn will show the Daily Imported Energy
[image]

SNAPE15, 2024-04-05:
> I use Total Imported Energy in the Energy Dashboard. This in turn will show the Daily Imported Energy Not quite the same but thats how I check how much I daily energy I imported/exported/generated/used/etc <img alt="image" width="356" src="https://private-user-images.githubusercontent.com/22875562/319838062-d71e3c6a-c038-4dea-8a22-fabbe2b34af6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIyODQ0NDIsIm5iZiI6MTcxMjI4NDE0MiwicGF0aCI6Ii8yMjg3NTU2Mi8zMTk4MzgwNjItZDcxZTNjNmEtYzAzOC00ZGVhLThhMjItZmFiYmUyYjM0YWY2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDAyMjkwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTk3OWIyODA2ZjI5NWYyYjQ4OWExOTI1YTRlMGQyNTliY2I4NjU4ODU5OWVmZTY5NTM5OWMxMTgyOTAyMzkyY2YmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.vTOKfBJdctHIC2wM-jmOcYM-UvT45tXEOsXJsx39hm8">

Currently I use the GoSungrow addon to use a few features that dont work via SHx integration as a work around. Its grabs the 'calculated' daily import from isolarcloud. Just hoping it could be done via modbus but doesnt seem like it.

housemaister, 2024-04-05:
You can easily define a helper in home assistant (utility meter) to calculate your daily imported energy (or any other interval or sensor). (see https://www.home-assistant.io/integrations/utility_meter/ for documentation)

NickStallman, 2024-04-13:
After extensive poking around with a range of options, I've determined the easiest way to get fast reliable data from my SH10RS is via the WiNet S2 websocket API.

I've mocked up some Typescript code to talk to the websocket API and extract all available information and feed it to Home Assistant with all the correct types and units every 10 seconds. With my SH10RS and SBR192 I've got both devices showing separately in Home Assistant with 83 sensors showing up between them.

Since it's a local API there's no cloud dependency and no issues with hammering it as much as you want (10 seconds seems to be a good rate), and it doesn't interfere with any normal operations either.

If there's interest I could wrap this up in to a HA addon. It's rather simple, only needing the WINet S2 IP/hostname and a MQTT connection string. The local API doesn't need any complicated authentication.

SNAPE15, 2024-04-13:
> After extensive poking around with a range of options, I've determined the easiest way to get fast reliable data from my SH10RS is via the WiNet S2 websocket API.
>
> I've mocked up some Typescript [...]

Does it allow for control of PV to Battery charge rate (power)? i.e. set battery to only charge at a rate of 2000W from PV.
Does it show all 4 MPPT 's?

danielclau, 2024-04-14:
> If there's interest I could wrap this up in to a HA addon. It's rather simple, only needing the WINet S2 IP/hostname and a MQTT connection string. The local API doesn't need any complicated authentication.

Yes please. If you've already reliably obtained the metrics, please share as quite a few of us here are still waiting for it.

NickStallman, 2024-05-05:
Ok I've published a new addon which gets all the data from the Winet S2 dongle over it's internal websocket API.

https://github.com/NickStallman/home-assistant-repo/tree/main/winet-extractor

Please treat this as alpha quality, however I've been running it for ~3 weeks now and I'm getting really good live data and it's quite stable.
Simply install the addon, then on the configuration page provide the Winet's IP and a MQTT url and that's all the required configuration done. It will automatically configure all the Home Assistant sensors that you can see under Devices -> MQTT and if you've got a inverter plus battery you'll see both devices separately.

Any feedback is appreciated.

jbardu, 2024-05-23:
> Ok I've published a new addon which gets all the data from the Winet S2 dongle over it's internal websocket API.

I'm a little confused over the status of RS inverters and access. I've got a new SH8.RS inverter with isolarcloud and it has no external lan port. So it fails the picture at the top of this GitHub. But it does listen on port 502 and I appear to be able to query slave 1 registers. Also when logging in to the inverter web interface I think it has modbus forwarding configuration tabs. 

So is this enough to get the integrations that would allow me to write some intelligent charge/discharging stuff? Or am I missing something -- Should I use your Winet S2 integration tool ?

thanks!

NickStallman, 2024-05-23:
@jbardu so to pull info from your inverter with frequent 10 second updates, it's best to use my addon.

Then take the mkaiser YAML file, and delete or comment out all the modbus sensors that are duplicates of what my addon provides.
They are redundant since my addon can get the data faster and more efficiently. This step seems to improve the stability of the Winet dongle which doesn't seem to like too much modbus traffic.

Then you can use the mkaiser settings and automations for control in your own automations. :)

After looking in to it further, my addon possibly wont get write support as it seems the Websockets API I'm using doesn't allow for changing values. For that there's a HTTP API and Modbus, and Modbus allows for a few extra settings which the dongle's UI doesn't show at all so it's probably best.

jbardu, 2024-05-24:
>  For that there's a HTTP API and Modbus, and Modbus allows for a few extra settings which the dongle's UI doesn't show at all so it's probably best.

thanks, any links to anything on the HTTP API ? you mean a REST interface or something less documented.

NickStallman, 2024-05-24:
It's an entirely undocumented Websocket API which is used when you navigate to the Winet's IP address directly. It's rather efficient and had all the values I needed so I reverse engineered it to use that interface to get the data in to Home Assistant.

Stephan-87, 2024-05-24:
2024-05-24 03:36:46 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized

jbardu, 2024-05-24:
> It's an entirely undocumented Websocket API which is used when you navigate to the Winet's IP address directly. It's rather efficient and had all the values I needed so I reverse engineered it to use that interface to get the data in to Home Assistant.

thanks, I'm still totally lost as to how programmatically people manage a battery dynamically. Or do they. With any sungrow inverters. Is it writing to certain modbus registers?

Stephan-87, 2024-05-24:
> It's an entirely undocumented Websocket API which is used when you navigate to the Winet's IP address directly. It's rather efficient and had all the values I needed so I reverse engineered it to use that interface to get the data in to Home Assistant.

2024-05-24 03:36:46 error: MQTT error: ErrorWithReasonCode: Connection refused: Not authorized 

any help please

NickStallman, 2024-05-24:
> thanks, I'm still totally lost as to how programmatically people manage a battery dynamically. Or do they. With any sungrow inverters. Is it writing to certain modbus registers?

mkaiser provides some automations/buttons/etc.. which you can then trigger with your own automations.
The "DefaultDashboard" shows all the values, knobs and buttons you can interact with and figure out your own automations to trigger them from there.

> any help please

For my addon it's best you open an issue there rather than clutter mkaiser's issues.
However you need a basic MQTT setup first like what's explained in https://haprofs.com/setting-up-mqtt-mosquitto-broker-home-assistant/
Do the "Install Mosquitto MQTT Broker on Home Assistant", "Install MQTT Broker via Home Assistant add-on" and "MQTT Authentication" steps, then provide the user/pass and ip to my addon in the mqtt url format.

Stephan-87, 2024-05-24:
> Install MQTT Broker

I have already installed MQTT Broker and work with Zigbee2MQTT and i have 7 devices work without issue 
Just question 
The MQTT URL is in the format of mqtt://<username>:<password>@<host>
can you write examples plz 

I write something like this 
mqtt://test:test@192.168.68.147 Is that correct ? Thanks for your help

SNAPE15, 2024-05-24:
> > Install MQTT Broker
> 
> I have already installed MQTT Broker and work with Zigbee2MQTT and i have 7 devices work without issue Just question The MQTT URL is in the format of mqtt://:@ can you write examples plz
> 
> I write something like this mqtt://test:test@192.168.68.147 Is that correct ? Thanks for your help

Need to create a user and passwd in mqtt.

ABredhauer, 2024-06-01:
> For anyone else with this model inverter I've found the cause of the remainder of the modbus errors I was getting. In the latest doc (above) there are a bunch of registers that are listed as not [...]

For those of us sticking with this integration I've found that the backup power values are now available following one of the recent firmware updates from Sungrow. Still to change for the MPPT3 and 4 readings though

frankyman88, 2024-06-08:
@ABredhauer I have just been testing with the new firmware too (updated inverter, communications dongle and battery this week). The max battery charging and discharging power registers (33047 and 33048) now work, and control charging speed even while on self consumption mode!! So I have switched all my automations from using forced charge speed to using the max battery charge speed instead :) will update with my experiences.... also with dynamic connection when it finally gets running (what a run-around, I'll update on the whirlpool about it more when it's all working).

frankyman88, 2024-06-08:
@NickStallman I am keen to try your Websocket API interface for pulling data from the inverter instead of Modbus, but I run home assistant in a docker container and therefore can't install addons. Any possibility of spinning up a separate docker container, proxmox LXC, HACS integration or other way of accessing this? Appreciate your efforts!!

SNAPE15, 2024-06-08:
> @ABredhauer I have just been testing with the new firmware too (updated inverter, communications dongle and battery this week). The max battery charging and discharging power registers (33047 and [...]

SH10RS inverter still?

ABredhauer, 2024-06-08:
>The max battery charging and discharging power registers (33047 and 33048) now work, and control charging speed even while on self consumption mode!! 

Sorry I completely forgot to mention those now work as well which was probably the bigger news. I'm still manually setting them while I work out how I want to automate them but they are working very well. Coming into winter has dropped my production but I still occasionally peak over the 10kw mark and can now use all my production

> 
> SH10RS inverter still?
Yes those comments were for the SH10RS

SNAPE15, 2024-06-09:
Yep, just tested after firmware update. Awesome. Going to make things a lot better over summer!

closed 2024-12-08 by github-actions[bot]