mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #189: Modbus stoped after a few hours

Cha40s, 2023-10-19:
**Describe the bug**
After a few hours the modbus connection stoped working. The last value of the sensors dosnt change anymore. 

**Your Sungrow inverter:**
 - Model: SH-10.RT v11
 - The inverter is connected via (mark one)
   - [] LAN (internal port)

- Are you using a Modbus Proxy (mark one)
  - [] no 

**Home Assistant :**
 - Version: 2023.10.3

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2023-09-29

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
Occurs after a few hours. 

```
Logger: homeassistant
Source: components/modbus/modbus.py:436
First occurred: 16:59:49 (18513 occurrences)
Last logged: 18:07:01

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/modbus/sensor.py", line 113, in async_update
    raw_result = await self._hub.async_pb_call(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 465, in async_pb_call
    result = await self.hass.async_add_executor_job(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/modbus/modbus.py", line 436, in pb_call
    result: ModbusResponse = entry.func(address, value, **kwargs)
                             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 117, in read_input_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 180, in execute
    return self.transaction.execute(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 190, in execute
    response, last_exception = self._retry_transaction(
                               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 268, in _retry_transaction
    return self._transact(packet, response_length, full=full)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 285, in _transact
    size = self._send(packet)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/transaction.py", line 325, in _send
    return self.client.framer.sendPacket(packet)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/framer/base.py", line 68, in sendPacket
    return self.client.send(message)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 173, in send
    if data := self._check_read_buffer():
               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/pymodbus/client/tcp.py", line 162, in _check_read_buffer
    ready = select.select([self.socket], [], [], end - time_)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: filedescriptor out of range in select()
```

mkaiser, 2023-10-23:
I honestly have no clue what could be wrong

I would try to use a modbus proxy, just because I don't know what else to try ...

Cha40s, 2023-10-26:
OK I try with modbus proxy. It worked for almost a year without one. I'll keep you informed.

dylan09, 2023-10-26:
Maybe you could check the HA ModBus documentation. There are a few changes with HA 2023.9 and 2023.10. Maybe you have to change the settings for error retries, error discovery, ...

If it started after the HA update to 2023.9 or 2023.10 I would give it a try before switching to a modbus proxy.

alexonbandit, 2023-10-31:
Hi, I do have exactly the same issue. Does anyone have new infos or better a solution?
THX

Cha40s, 2023-10-31:
Modbus Proxy solved my issue. Took me 10 Minutes to set up.

alexonbandit, 2023-10-31:
Ok thanks for that hint.
Never did this before. May I kindly ask you @Cha40s to give me some advice how to set it up?
Many thanks in advance!

Cha40s, 2023-10-31:
Sure I use this add on for home assistant. https://github.com/Akulatraxas/ha-modbusproxy/
it’s outdated but still works. Just install it and change the ip in the add on configuration tab to your Sungrows ip. After that change the ip for the mqtt configuration in the sugrow yaml to 127.0.0.1. If you want to use an additional program like Evcc just  get the data from the proxy by using the ha ip with port 502.

alexonbandit, 2023-10-31:
Thanks a lot! I will give it try and yes I am also using EVCC 😉 so also great that you anticipated that already

alexonbandit, 2023-11-01:
@Cha40s thanks again I got it working!
For now using the proxy seems indeed to have a more stable connection.
Fingers crossed 🤞