mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #202: Support for Register 5113 and 5006

bleughb, 2023-11-06:
Hey,
Owner of two SG5.0RS inverters here, 
the code's working superbly with both of my inverters, didn't take long to do a simple name change for each sensor to add a number 2 to the end!
I'm trying to modify the YAML to add the following registers 
5113 - Daily running time
5006 - Start_Stop

But, I don't get a sensible answer out of 5113 - the response is always '6'
if I query the registers with 'sungather' - I get sensible responses out of them, and also out of 5038 - work state 1 (which is always 0 through this code)

Any advice on how to change things?

Louisbertelsmann, 2023-11-07:
Is there something wrong with the datatype?

bleughb, 2023-11-08:
Nope, should just be uint16 - I've checked a few other repositories - sungather for instance works and reports the correct run time for 5113....BUT, I can't use Sungather as I have TWO inverters...The method here, having two YAML files works superbly well

bleughb, 2023-11-08:
Ok, 'resolved' in a fashion.
I tested the data coming from the inverter with a TCP modbus reader (Schneider had a great one)....Reading 5113 there gives the correct data in minutes....

Tried reading a bunch of registers from 5100 upwards and noticed
There's some type of weird offset

register 5113 seems to be reading actual register 5114 which is Country - 06 Australia
so, I tried 'reading' register 5112  - this gives the actual runtime correctly

      - name: Daily Operation Time
        unique_id: sg_daily_operation_time
        slave: !secret sungrow_modbus_slave
        unit_of_measurement: min
        address: 5112 # this should be 5113 but is offset by -1 for some reason
        input_type: input
        device_class: duration
        state_class: measurement
        data_type: uint16
        scan_interval: 60

bleughb, 2023-11-08:
Closed, Modbus Register 5113 (daily running time in minutes, resets to 0 every day) actually reads on 5112 for some reason.
Modbus register 5006 is a U32 data type for total running time for later inverters.

closed 2023-11-08 by bleughb

Louisbertelsmann, 2023-11-08:
The other registers also have that offset. It's normal.

bleughb, 2023-11-08:
> The other registers also have that offset. It's normal.

Where does the offset start though?
some have an offset, some don't?
is something documented somewhere about this?
it's not obviously seen in some other modbus inverter applications, 

Just curious now and it'll save me an evening of tinkering if it's already been covered elsewhere