mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #206: Yearly PV generations jumps around

majuss, 2023-11-08:
**Describe the bug**
A clear and concise description of what the bug is.

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT v112]
 - The inverter is connected via (mark one)
   - [] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [] I don't know what that is

**Home Assistant :**
 - Version: [e.g. 2023.09.1]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file)

** Inverter Firmware Status:**
   - [] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]
**Additional context**
Add any other context about the problem here.

poehlerflorian, 2024-04-03:
I have the same problem. Got my PV System connected in March. Yearly PV Generation (Current and 2024) value was correct until 2024-03-31 23:59. The first value in April jumped to the 1,8x of the correct value. Same thing with Yearly Export, but this value jumped to the PV Generation value in April.

I am not sure if this is an error of the HA Integration or if the inverter transmits wrong values.

housemaister, 2024-04-03:
Same here. It seems to be a bug at Sungrow... also there are some reports (dated 24th December) in this thread in a German speaking forum: https://www.photovoltaikforum.com/thread/166134-daten-lesen-vom-sungrow-wechselrichtern-modbus/?postID=3528584&highlight=6265#post3528584

I would suggest not to rely on the yearly values. Instead generate you own with helpers in HA.

poehlerflorian, 2024-04-18:
I now noticed that the values have jumped back down again and seem to be correct. Since the 16th of April "Yearly export (current)" e.g. shows the correct value.

The Yearly values of 2024 and current are now the same and also have a <1 kWh difference to the total values.

I guess, Sungrow probably fixed the values provided by the inverter.

poehlerflorian, 2024-04-19:
> > I now noticed that the values have jumped back down again and seem to be correct. Since the 16th of April "Yearly export (current)" e.g. shows the correct value.
> > The Yearly values of 2024 and current are now the same and also have a <1 kWh difference to the total values.
> > I guess, Sungrow probably fixed the values provided by the inverter.
> 
> Did you do some update of the inverter firmware? Seems strange that Sungrow can fix this without update.

Good point, no I did not.. I did however update to HA 2024.4 a day prior and revert back to 2024.3 (since I had missing values like in #274) the day after that. After reverting, I got the correct values again.

[image]

mkaiser, 2024-05-06:
can also confirm...
[image]

The "real" export is around 950kWh this year.  I guess the spikes at the end of the month causes this. Roughly adding the "linear increases" would match the right values. 

These are non-documented registers by sungrow.  It is a firmware-thing, so there is no way to fix this within this integration. 

I am thinking of completely removing them, as they don't bring any benefit, compared with the HA energy dashbord

poehlerflorian, 2024-05-06:
On May, 1st, the values jumped up again for me.

I agree with you that the values don't bring any benefit, and I think I would also remove them.

What I find strange however is that the values got fixed for me in April after reverting an HA update.. If the values are simply read from the register, an HA update should not affect them in any way.

Louisbertelsmann, 2024-05-06:
> These are non-documented registers by sungrow. It is a firmware-thing, so there is no way to fix this within this integration.
The sensors aren‘t undocumented. They are in the normal register document.

closed 2024-11-17 by github-actions[bot]