mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #208: Daily load consumption?

elektrinis, 2023-11-12:
We have daily import and export, daily battery charge and discharge, daily PV generation... One thing I can't find is daily load consumption, basically I have no idea how much my house has consumed daily and overall?
I would love to have this number so I could list all other consumers and see what's missing. This should look awesome on [Sankey card](https://github.com/MindFreeze/ha-sankey-chart).
[image]

mkaiser, 2023-11-12:
Integrated it some days ago :) 

Check https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/62be4a69a8f20098c324147e731e2560634ce1b4/modbus_sungrow.yaml#L2238

[image]

mkaiser, 2023-11-12:
How many energy meters do you have installed? looks very fine-grained :)

dylan09, 2023-11-13:
> Integrated it some days ago :)
> 
> Check
> 
> https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/62be4a69a8f20098c324147e731e2560634ce1b4/modbus_sungrow.yaml#L2238

Is the sensor `Daily consumed energy` working without spikes? When displaying as graph I have a lot of spikes. I think they are related to timing issues when they are delivered by modulus integration.
I have changed it to a trigger template. Time pattern. Calculated every 55 seconds. Since I have changed this, they are mostly gone. 

Looks like this. And this sample only shows small spikes. Sometimes they was very heavy. 
[image]

mkaiser, 2023-11-13:
in my installation it does not "ripple" as in your screenshot. But I had an "undershoot" at midnight.

[image]

I guess we can overcome this by using a lopass filter:

https://www.home-assistant.io/integrations/filter/#low-pass

```
    filters:
      - filter: outlier
        window_size: 4
        radius: 4.0
      - filter: lowpass
        time_constant: 10
        precision: 2
```

(values are from the HA doc, did not test or change)

The energy sensors are read every 10min. 
So maybe a large window size would be good

I am not at home at the moment. Could someone test?

dylan09, 2023-11-13:
I have added a new filtered sensor to test If filtering will work.

dylan09, 2023-11-16:
As expected. The filter will smooth to much.
I tried:
```
  filters:
    - filter: outlier
      window_size: 4
      radius: 4.0
    - filter: lowpass
      time_constant: 5
      precision: 2
```

First the graph without the filter (I use a `time_pattern` trigger every 55 seconds.
[image]

Second the graph from sensor with filter applied:
[image]

mastameista, 2023-12-19:
I have some log entries (level warning) from recorder integration:

```
Logger: homeassistant.components.sensor.recorder
Source: components/sensor/recorder.py:288
Integration: Sensor ([documentation](https://www.home-assistant.io/integrations/sensor), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+sensor%22))
First occurred: 12:15:10 PM (1 occurrences)
Last logged: 12:15:10 PM
Entity sensor.daily_consumed_energy from integration template has state class total_increasing, but its state is not strictly increasing. Triggered by state 17 (18.0) with last_updated set to 2023-12-19T11:12:36.581022+00:00. 
```
The time relates to a point short before the "ripple". 

[image]

And I also have some "underruns" almost every night.

mkaiser, 2023-12-19:
oh, almost forgot this issue ;)

I tried another filter, which seems promising. 

Add this just above 
```
# 'virtual' template sensors for better readability
template:
```
... (left aligned)

```
sensor:
  - platform: filter
    name: "Daily consumed energy (filtered)"
    entity_id: sensor.daily_consumed_energy
    filters:
      - filter: time_simple_moving_average
        window_size: "00:05"
        precision: 2
```

[image]

Can someone with "some sun" confirm? Over the last dark days I had only around 1 kWh/day yield....

dylan09, 2023-12-20:
Put it on my list for next year. Currently no sun also.  I think my test will start mid of january.

mastameista, 2023-12-21:
@mkaiser Thx! I've added the new sensor. We had some nice sunny days this week but now its bad weather until Monday. Will update here after some measuring in the next weeks.

mastameista, 2024-01-19:
@mkaiser 
Filtered version looks good! There's just one "glitch" I can't explain. But it's definitely not systematic anymore. 

[image]

closed 2024-11-17 by github-actions[bot]