mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #210: min_soc and 2 others entities are not available anymore

iBLtz, 2023-11-13:
**Your Sungrow inverter:**
 - Model:  SH-10.RT v112
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant :**
 - Version: 2023.11.2

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2023-11-12 

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

Hi all,

since a couple of weeks, I cannot see the min_soc value anymore. It used to work and i do not remember when it stopped working. 
The cursor to change the min_soc is working properly. When  I set something there, I can see the value in iSolarcloud.

[image]

There are 2 more entities which are not available (see screenshot)

Any idea what i have to do to correct this behavior?

Thanks in advance

mastameista, 2023-11-14:
Strange, I have the exact same setup and it's working. Re Firmware of inverter:
[image]

Could be different depending in which country your solar plant is installed.

Maybe copy/paste the modbus_sungrow.yaml again?

mkaiser, 2023-11-14:
Lets check, if it is a modbus/sensor issue or an issue with the visualization:

Navigate to
Developer tools --> States 

and use the "filter entities" to look for "set_sg_inverter_run_mode". It should look like this:

[image]

If "set_sg_inverter_run_mode" is like in the screenshot, you should use the current dashboard.yaml from the repo

iBLtz, 2023-11-14:
The entity is available like in your screenshot. I updated to the current dashboard.yaml and the inverter mode is there again : 
[image]

Unfortunately, sensor.min_soc is still not available. Here is a screenshot when I filter with min_soc 

[image]

iBLtz, 2023-11-14:
> Strange, I have the exact same setup and it's working. Re Firmware of inverter: [image]
> 
> Could be different depending in which country your solar plant is installed.
> 
> Maybe copy/paste the modbus_sungrow.yaml again?
[image]

I updated it yesterday to be sure it is the latest version

mkaiser, 2023-11-14:
Is something modbus-related in the logs?

Settings --> System --> Logs (i guess in german it is "Protokoll", and not the "Logbuch" )

iBLtz, 2023-11-14:
no modbus error in the logs :( 

[image]

I wonder if the entity is somewhere store on the database and somehow corrupted?

mastameista, 2023-11-14:
you could check /config/.storage/core.entity_registry and search for "sensor.min_soc" -> this is the HA side of the story. 
But never had an issue with this config / registry. 

If there's no modbus error in your log file it probably makes no sense to check the register according to modbus_sungrow.yaml:
address: 13058 # reg 13059

Last resort: Restore a backup to a point where it was working and test. You mentioned "since a couple of weeks" and that you have 2023.11.2 installed. This version of HA was released some days ago. So i strongly assume that it didn't work before either. 
Reboot your inverter?

closed 2023-11-15 by iBLtz