mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #213: max SoC value for Backup from 75% to 90%

ThLehmann, 2023-11-19:
for balancing the battery in winter it is necessary to keep the battery at a high level. For this, a backup SoC of up to 90% would be desirable. There is currently a limit of 75%

mkaiser, 2023-11-19:
Hi, 

datasheet says 
[image]

currently the range of the input selector is limiting to 75%

```  set_sg_reserved_soc_for_backup:
    name: Set reserved SoC for backup
    min: 0
    max: 75
    step: 1
``` 

I can't remember the reason why we set it to 75%.... (maybe it is something Sungrow changed...)

Can you change the parameter to 100 and give me a feedback, if it works? (Can't test myself - now backup and pretty much no sun the last weeks...)

If it works as expected, I am happy to integrate it :)

ThLehmann, 2023-11-19:
Hey, perfect ! yes the 100% works as well
but it make not really sense in my opinion, 90% is enough but if 100% is possible its the choise of the user ;)

[image]

mkaiser, 2023-11-20:
done - thank you :)

closed 2023-11-20 by mkaiser