mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #217: Extremely high values distort the statistics

dvogt23, 2023-11-29:
**Describe the bug**
Some days I got weird values in my statistics, but no idea why. Have to manuelly correct them in statistics sections, to get my ~ real values in energy.

**Your Sungrow inverter:**
 - Model: SH10RT-V112
 - The inverter is connected via (mark one)
   - [X] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [X] no 
  - [] I don't know what that is

**Home Assistant :**
 - Version: 2023.11.3

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2023-09-29

** Inverter Firmware Status:**
   - [X] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.

Louisbertelsmann, 2023-11-29:
Update to the newest version. There was an issue with earlier versions.

dvogt23, 2023-11-29:
Thanks, will do. Lets see...

closed 2023-11-29 by dvogt23