mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #221: Next version to extend support: Help needed to create a Python-based yaml-generator

mkaiser, 2023-12-01:
Hey, 

**@all Python-affine programmers and others willing to help :)**

some time ago I made some experiments to keep up with the latest developments of new sungrow models and their different feature levels (e.g. WiNet-S attached inverters do not support history values, RS inverters have a very different register mapping). Additionally, supporting multiple inverters requires manual modifications. 

I still like the idea of having a single configuration file (modbus_sungrow.yaml), but with the limitations of yaml (no globally set variables (e.g., for the inverter model) and no if/then/else for sensors, this won't be possible :/

As a workaround I started to create a YAML-file-generator based on Python and received some great help from Whirl (discord). 

**There are still some issues with this generator and due to limited time I need your help:** 

The YAML generator (https://github.com/mkaiser/modbus_sungrow_yaml_generator) works this way:
- An extended .yaml file is used as input for the parser and contains annotated sensors / templates / automations of different inverters  and connections (ETH vs. WiNetS)
- The yaml generator reads the annotations and creates separate yaml-configurations, e.g. 
  - modbus_sungrow_1_inv_SHxRT_basic_WiNet_WLAN
  - modbus_sungrow_1_inv_SHxRT_all_ETH
  - modbus_sungrow_2_inv_SHxRT_basic_battery_WiNet_WLAN.yaml
  - modbus_sungrow_2_inv_SHxRT_sungrow_battery_WiNet_WLAN.yaml
- currently forseen annotations:
  - inverter: '1', '2'
  - model: 'SHxRT', 'SHxRS', 'SG', 'all': 
  - level: 'basic', 'extended', 'battery'
  - connection: 'Eth', 'WiNet-WLAN', 'WiNet-Eth'
- Github actions are used to automatically generate and release several yaml-configurations (never did this before, maybe someone has already some matching code snippets?)

The current version of the generator is in the separate github repo. Whirl (discord) helped to get the filtering done. My (still limited) Python skills and spare time are not enough to implement that in the near future...

**Current blocker:** 
- comments in the original file are not preserved (we want to keep the header file)
- annotations (mode, level, connection) of not-filtered sensors need to be commented to be parse-able by home assistant

**It would be very great for a lot of people, if someone could help us :)**

Some issues / discussions of new sensors, which would benefit from this parser: 
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/211 (Wallbox)
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/220 (SH5K WiNetS)
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/discussions/194 (2nd inverter)
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/209 (2nd inverter)
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/43 (2nd inverter)
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/185 (meter voltage / only LAN?)
https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/184 (SH10RS)
grid frequencies of WiNetS and ETH differ by factor 10, but both use the same register address

**Willing to help?**
--> Suggestions are welcome everywhere: in this issue, discord, the generator repo https://github.com/mkaiser/modbus_sungrow_yaml_generator or whatever :)

~Martin

morph027, 2023-12-10:
Looks like jinja templates might help to create files including logic. Will try to have a look.

mkaiser, 2023-12-13:
> I love this idea and happy to help. I have been following the discussion on the development of the generator and I am wondering if maybe a slightly different input file format would work? The issue I see is that the current input file is a mix of HA's yaml format with some extra elements that help you work out how to parse it. I'm wondering if the input file should abandon the HA format and become more of a list of registers with attributes to define the sensor type and compatible inverters, connection methods etc. It would mean the generator would probably have to become more complex to build the HA format but it may also be easier to maintain?

yes, in some ways you are right. But the parser is already working for the HA format. Only the comment handling is an open issue. 
Having one file for register / model settings only would require more addional python work on top of this (check first which model, then load supported registers and add more registers, if a battery is detected, add suitable automations) This would be great for a future native HACS / HA integration. I would love to support this, but more with my knowledge on the sungrow modbus register / configuration stuff, not implementing it (no spare time :/ )

> Looks like jinja templates might help to create files including logic. Will try to have a look.

What do you have in mind? Could you give me a link?

morph027, 2023-12-14:
> > Looks like jinja templates might help to create files including logic. Will try to have a look.
> 
> What do you have in mind? Could you give me a link?

Did not had the time to look at the generator....do you have an example problem you like to solve for me?

Like:

Create this block of yaml for 'Eth', 'WiNet-WLAN', 'WiNet-Eth' .... Or so?

mkaiser, 2023-12-19:
In the file  ``` [input_minimal.yaml](https://github.com/mkaiser/modbus_sungrow_yaml_generator/blob/main/input_minimal.yaml) ```

there are several possible variations and HA sections like in the origininal modbus_sungrow.yaml 

At the end of the yaml-generator.py there are some use cases. 

I try to compile a complete input file over the holidays, but the minimal file should be a good start to fix the comment issues

bagbag, 2024-01-04:
How about a simple HTML/JS frontend, which can be hosted as a static file over GitHub Pages?

Instead of a yaml template, use a config json that contains all the information required to create the yaml file. This way you don't have to redo the whole template and instead just adjust the generator.

I'd be willing to help out with that, but I need more information about how all this stuff is linked together, as I am pretty new to HA and failed adjusting the yaml for a 2nd sungrow unit.

mkaiser, 2024-11-11:
I guess the most promsing approach is currently writing a "native" Homeassistant integration. Please check 

https://github.com/AlexanderLanin/homeassistant-sungrow/

closed 2025-03-18 by github-actions[bot]