mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #226: Battery temp from Battery

kalaws, 2023-12-20:
Right now we are getting Battery temp from Inverter. However, at least in iSolarCloud battery temp is also reported by the battery. The difference is that Inverter gives no decimals, but rounds to integers. Batttery gives a precison of one decimal. Any way we could get this (hopefully through WiNet-S in my case, unfortunately)?

dylan09, 2023-12-20:
Scale for the sensor is defined as `0.1`. So it should have decimals. But it seems the inverter only reports temperature without decimals. Also on my LAN port.
But how accurate is the measured value? Does the precision of the measurement justify decimals?

kalaws, 2023-12-20:
Yes, in iSolarCloud one can see 20,7, 20,8, 20,9 etc.

kalaws, 2023-12-20:
And there is also in iSolarCloud the inverter value, which at the same time simply shows 21.

frankyman88, 2024-01-04:
My Battery temperature entity in home assistant reports one decimal place just as in ISolarCloud.
In your modbus YAML make sure you have the following line in the battery temperature sensor config
`precision: 1`

closed 2024-05-06 by mkaiser