mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #228: 2024.1

Mats1717, 2023-12-28:
**Describe the bug**
Tried 2024.1 now and its working but gives alarm that some code will be removed if not it will stop working later on.
"retries and swap" i think it was

dewenni, 2023-12-31:
yes, there is a change regarding the `swap` parameter.
[image]

The parameter swap itself will be still valid. But the argument none was removed.
You should only use swap parameter if you have to swap bytes.

[image]

dewenni, 2023-12-31:
and the parameter `retries` seems to be completely removed

[image]

mkaiser, 2023-12-31:
Thank.you for the hints. Will make the updates, soon!

mkaiser, 2023-12-31:
Just checked. The "swap: none" was already removed some time ago. 

I just removed the remaining "retries" in the repo, and updated to 2024.1 -> No warnings :)

closed 2023-12-31 by mkaiser