mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #229: Error: device: 1 address: 13050 -> Modbus Error: [Input/Output] [Errno 32] Broken pipe

smartmatic, 2024-01-05:
**Describe the bug**
HA Log throws errors and entities are randomly not available for some seconds. Then they appear again.

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT v112] 
 SH8.0RT-V112

 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [x] I don't know what that is

**Home Assistant :**
 - Version: [e.g. 2023.09.1] 
 2024.1.0

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file) 
2023-12-31

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
No errors like provided and entities should constantly available.

**Screenshots**
[image]

**Additional context**
Behaviour that entities are shortly and randomly not available appears since HA update to 2024.1.0

NBD99, 2024-01-31:
I can confirm i have the same issue on a SH10RT inverter.

umrath, 2024-05-21:
Since I've moved to modbus proxy I started to see similar effects.
My battery related entity vanished while the rest stays available.

closed 2024-11-03 by github-actions[bot]

umrath, 2024-11-03:
The problem persists.

Gnarfoz, 2024-11-04:
Communication errors can't really be fixed here. This package does no communication by itself. It's "merely" a big configuration file for Home Assistant's built-in Modbus functionality.
That said, are you sure Home Assistant is the only thing accessing the inverter via Modbus?
The Modbus stack on the inverter is notoriously unstable when multiple clients try to access it.