mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #234: retries` configuration key is being removed

JensPohl69, 2024-01-08:
**Describe the bug**
A clear and concise description of what the bug is.

**Your Sungrow inverter:**
 - Model:  SH-10.RT v112
 - The inverter is connected via (mark one)
   - [] LAN (internal port)
   
- Are you using a Modbus Proxy (mark one)
      - [] I don't know what that is

**Home Assistant :**
Core 2024.1.0
Supervisor 2023.12.0
Operating System 11.2
Frontend 20240103.3

**modbus_sungrow.yaml:**
 - Version/ time stamp : 
 
 # Home Assistant Sungrow inverter integration
# https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant
# by Martin Kaiser
# last update: 2023-12-02

# Note: This YAML file will only work with Home Assistant >= 2023.10

** Inverter Firmware Status:**
   - latest

Since 4 days i get the following warning from the system. No update by now. 
# 1 repair
# `retries` configuration key is being removed
# This stops working in version 2024.7.0. Please address before upgrading.
# Please remove the retries key from the modbus entry in your configuration.yaml file and restart Home Assistant to fix this issue.
 # The maximum number of retries is now fixed to 3.

Result by checking the configuration in the developer tools 

Configuration warnings
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_host_ip' is an invalid option for 'scene', check: states->0->sungrow_modbus_host_ip
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_port' is an invalid option for 'scene', check: states->0->sungrow_modbus_port
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_slave' is an invalid option for 'scene', check: states->0->sungrow_modbus_slave
Invalid config for 'scene' at configuration.yaml, line 15: required key 'entities' not provided
Invalid config for 'scene' at configuration.yaml, line 15: required key 'name' not provided

In line 15 of the configuration.yaml you find

scene: !include scenes.yaml 

This is the line you recomend in your manual

The data exchange with the sungrow converter is still running fine. 
I'm afraid when I do the pending upgrade it won't work anymore. 
since 4 days i get the following warning from the system. No update by now.

# 1 repair
# `retries` configuration key is being removed
# This stops working in version 2024.7.0. Please address before upgrading.
# Please remove the retries key from the modbus entry in your configuration.yaml file and restart Home Assistant to fix this issue.
 # The maximum number of retries is now fixed to 3.

Result by checking the configuration in the developer tools 

Configuration warnings
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_host_ip' is an invalid option for 'scene', check: states->0->sungrow_modbus_host_ip
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_port' is an invalid option for 'scene', check: states->0->sungrow_modbus_port
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_slave' is an invalid option for 'scene', check: states->0->sungrow_modbus_slave
Invalid config for 'scene' at configuration.yaml, line 15: required key 'entities' not provided
Invalid config for 'scene' at configuration.yaml, line 15: required key 'name' not provided

In line 15 of the configuration.yaml you find
scene: !include scenes.yaml  

This is the line you recommend in your manual
The data exchange with the sungrow converter is still running fine. 
I'm afraid when I do the pending upgrade

Thanks in advanced 

Jens

Louisbertelsmann, 2024-01-08:
Update to the newest yaml.

JensPohl69, 2024-01-08:
Got it. Thanks

JensPohl69, 2024-01-08:
Thanks a lot and I got it and reworked the yaml. Was easy.

But in the developer tools after check configuration the following warnigs are still exisiting.

Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_host_ip' is an invalid option for 'scene', check: states->0->sungrow_modbus_host_ip
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_port' is an invalid option for 'scene', check: states->0->sungrow_modbus_port
Invalid config for 'scene' at configuration.yaml, line 15: 'sungrow_modbus_slave' is an invalid option for 'scene', check: states->0->sungrow_modbus_slave
Invalid config for 'scene' at configuration.yaml, line 15: required key 'entities' not provided
Invalid config for 'scene' at configuration.yaml, line 15: required key 'name' not provided

Do you have any ideas what else I can do?
I suspect this warning is older.

Thanks Jens

Louisbertelsmann, 2024-01-09:
There’s something wrong with your installation. THis integration doesn’t use the scene.yaml.

JensPohl69, 2024-01-09:
Hi, I have removed the entries from Sense.yaml and restarted HA. The Integration runs as ususal and the warning is gone. Many thanks for your support. Kind regards JensVon meinem iPhone gesendet

Louisbertelsmann, 2024-01-09:
So, I can close this issue.

closed 2024-01-09 by Louisbertelsmann

JensPohl69, 2024-01-09:
Yes 🙌 Von meinem iPhone gesendet