mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #236: Battery Capacity is "0.00kWh"

mathiasolofsson87, 2024-01-16:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

Value for battery capacity is not existing in HA(battery_capacity).
[image]

In my Sungrow Plant Configuration on the iSolarapp the battery capacity is shown as 9.6kWh.

I have a SH6.0RT-V112 inverter with a 9.6kWh sungrow battery.

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT v112]
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [x] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2024.1.3]

**modbus_sungrow.yaml:**
 - Version/ time stamp : # last update: 2023-12-31
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

liquidblack, 2024-01-17:
Almost identical for me with a SH10RT v112. Fully patched. What the register that the capacity is read from?

liquidblack, 2024-01-18:
Reference to https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/184#issuecomment-1807053291 I found and tested that the capacity can be found in input register 13038 - the following works after a restart of home assistant

```      
        - name: Battery capacity
        unique_id: sg_battery_capacity
        slave: !secret sungrow_modbus_slave
        address: 13038 
        input_type: input 
        data_type: uint16
        unit_of_measurement: kWh
        device_class: energy
        scale: 0.1
        scan_interval: 10
```

I've never made a PR before but would love to contribute if possible :)

mathiasolofsson, 2024-01-19:
Unfortunately 13038 was not the adress for my battery_capacity on my inverter.

Also my battery_capacity info in the modbus_sungrow.yaml has input_type: "holding", not "input"

mastameista, 2024-01-19:
@mathiasolofsson, @liquidblack  you can set the capacity in iSolarCloud. There are two settings and under plant configuration is not enough. you have to set it in advanced settings too. For this you nid an installer account on your plant. 
HowTo: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/88#issuecomment-1646586986

btota, 2024-01-20:
@liquidblack works perfect with SH10RT-V11.2, thx!
@ll with setting up on ISC, I had an issue, which had an impact on my max_charge/discharge_power, I wasn´t able to charge/discharge more than 5 kW from battery. So I had to reset back to NULL on ISC. Also the work of [@alex&@alan https://github.com/AlexanderLanin/homeassistant-sungrow](url) showed me that capacity is working with HA.
@mkaiser @liquidblack PR would be great!

liquidblack, 2024-01-22:
> @mathiasolofsson, @liquidblack you can set the capacity in iSolarCloud. There are two settings and under plant configuration is not enough. you have to set it in advanced settings too. For this you nid an installer account on your plant. HowTo: [#88 (comment)](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/88#issuecomment-1646586986)

Thanks @mathiasolofsson - it was a bit of dance to get the account set up - for anyone else reading this I needed to added a separate account, create the account, remove the share, re-add the share and then it showed up.  I'll do some testing to see if the information shows up on the battery.

My open question then becomes - which value should be used if the system "knows" that it's a 9.6kWh battery but I have to set it in another place (to whatever I like) - shouldn't the inbuilt value be the one that's used?

gsteger81, 2024-06-18:
Hello,

I had the same issue, Battery capacity showed 0 in Dashboard. Setting the right Value in iSolarcloud with installer account like @mastameista suggested worked. [](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/236#issuecomment-1900403662)

purcell-lab, 2024-09-30:
@RafAustralia has some alternative registers to try for battery capacity at #354

closed 2025-02-04 by github-actions[bot]