mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #239: rapid reading of grid power

elektrinis, 2024-01-21:
I want to implement my own energy management system, and for that I need to read grid power quite frequently, more than once per second. Is this possible? All other values can remain slow, like they are now.

Louisbertelsmann, 2024-01-21:
Unfortunately, Home Assistant doesn't support readings faster than once a second. It would only be possible outside Homeassistant or using a python integration, but I am no expert at that. Also, if you go far enough, you might also overwhelm the inverter.

Louisbertelsmann, 2024-01-24:
I am using 1s refresh times on my inverter and it actually works better than 10s (Sungrow is weird, so the inverter basically closes the connection and isn't ready when modbus tries to read it...)
You would change the scan_interval of the sensors you want to read faster to 1 or something like that.

elektrinis, 2024-04-17:
Just confirming that I have changed the time to 1 second in yaml and it works. Would love to get this more frequent though.

closed 2024-04-17 by elektrinis

Louisbertelsmann, 2024-04-17:
Unfortunately you can't change the delay to a lower one. For that one would need to use pymodbus directly, which we would have to write an integration for.