mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #241: SH5k-30 issues

TaylorWoods1, 2024-01-23:
Hi All,

seeking some help on integration with SH5K-30 inverter. it seems that writing parameters is ineffective. written parameters remain from anywhere between a few seconds, and i have even seen up to a day on some markers, however they are eventually overwritten once again by the inverter. examples are:
battery_capacity = 655.35 kWh
ems_mode_selection = self-consumption mode (default)
battery_forced_charge_discharge_cmd = stop (default)

no matter what i do, these will always end up back at these values after a period of time.

Im assuming based on this, that the integration with SH5k-30 is not great and writing parameters is useless?
am i missing something?

second issue is that the battery SOC on the inverter is so unrealistic and it is forcing a CV charge state when the inverter thinks the battery is approaching the max battery voltage, yet my BMS is reporting a completely different SOC value. is there a way to force sync these on HA?

btota, 2024-01-23:
> battery_capacity = 655.35 kWh
seems register adress is wrong 13038 (13039 -1)

do you use newest *.yaml from mkaiser? HA uptodate? modbus-proxy?
can you check your modbus_sungrow.yaml, beneath you'll find my conf for Sungrow SH10RT-V112 with SBR 128

`- name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 13038 # reg 13039
        #address: 33048 # reg 33049
        input_type: input
        #input_type: holding
        data_type: uint16
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.1
        scan_interval: 600`

TaylorWoods1, 2024-01-23:
thanks for the response @btota. im using mkaiser's latest version from 3 weeks ago, dated 31-12-23.
home assistant is Core 2024.1.3, Supervisor 2023.12.1, Operating System 11.4, Frontend 20240104.0

      - name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 33048 # reg 33049
        input_type: holding
        data_type: uint16
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.01
        scan_interval: 600

TaylorWoods1, 2024-01-23:
sungrow_modbus_host_ip: 192.168.86.61
sungrow_modbus_port: 502
sungrow_modbus_slave: 1

TaylorWoods1, 2024-01-24:
@btota where are you getting your register addresses from?

btota, 2024-01-25:
[TI_20231019_Communication.Protocol.of.Residential.Hybrid.Inverter_V1.1.2_EN.pdf](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/files/14056006/TI_20231019_Communication.Protocol.of.Residential.Hybrid.Inverter_V1.1.2_EN.pdf)

heino42, 2024-10-05:
I just re-deployed the sungrow configuration from Github (with the new register 13038), result: 655,35kWh. Obviously wrong, looks like max value for 2 bytes.
Then I found another line in the technical document above:
[image]
40 Battery Capacity High precision 5639 U16 0.01KWh

And surprise, this one works: 10,2 kWh
Config used:
      - name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 5638 #33048 #13038 # reg 13039
        input_type: input # holding
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.01
        scan_interval: 60

Is this device-dependent? I'm using a SH8.0RT-V112, inverter update from 2023-08 "SAPPHIRE-H_03011.51.03".

Gnarfoz, 2024-10-08:
The document says register 13039 / address 13038 is only for SH_RS and SH_RT, so it's not surprising it doesn't work for @TaylorWoods1's SH5K-30.
[image]

Are you connecting to the WiNet-S inteface or the inverter's own LAN port, @heino42?

Either way, if 5639 / 5638 works for _all_ hybrid inverters, then both 13039 and the undocumented 33048 should be obsolete and be replaced.

heino42, 2024-10-09:
> Are you connecting to the WiNet-S inteface or the inverter's own LAN port, @heino42?
LAN port for Home Assistant, WiNet-S for iSolarCloud
> 
> Either way, if 5639 / 5638 works for _all_ hybrid inverters, then both 13039 and the undocumented 33048 should be obsolete and be replaced.
Agree - would be great to have a working solution for all models in scope!

closed 2025-02-11 by github-actions[bot]