mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #242: Add information of EV charger (for car) to available values

GolfOWL, 2024-01-26:
First I want to thank all of you, especially mkaiser, for this great add-on in HA. I have had a PV system since yesterday and the add-on has made it very easy for me to integrate it into HA.

In addition to the inverter and the battery, I also have a wallbox from Sungrow, which is connected and controlled via the management system.

As a result, I can imagine that corresponding parameters are also available for the "EV Charger1", which can be integrated into HA.

I would like to address such an extension with this feature request.

I would like to thank you in advance for your feedback and hope that I have done this "correctly" in the form of this request.

andi-blafasl, 2024-09-23:
I have ordered the Wallbox yesterday and also want to integrate it into homeassisant via modbus. Found something about the registers in the [iobroker forum](https://forum.iobroker.net/topic/73087/sungrow-wallbox-ac011e-01-erfolgreich-mit-modbus-eingebunden/11). Maybe I get the wallbox up and running next weekend. Have some challenges with cabling from my house to the garage ;-) Hopefully the RS485 over Ethernet adapter will save me from pulling a new cable...

lassej01, 2024-09-24:
> I have ordered the Wallbox yesterday and also want to integrate it into homeassisant via modbus. Found something about the registers in the [iobroker [...]

Maybe you want it all local, but otherwise, you can use this already existing integration for wallbox in Home assistant:
https://www.home-assistant.io/integrations/wallbox/

Gnarfoz, 2024-09-24:
I doubt that works with the Sungrow AC011E-01 wallbox.

Louisbertelsmann, 2024-09-24:
How about this? Disclaimer: I don’t have a wallbox, but other people have debugged it and it should work.
https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant

Louisbertelsmann, 2024-09-24:
If you want to read the wallbox through the inverter, you should be able to copy the sensors into the sungrow_modbus.yaml because sungrow doesn’t like it if two modbus hubs are on the same IP.

andi-blafasl, 2024-09-24:
Louis, that's a great find! Bit I think you do not need to integrate it with the inverter yaml. They are reading the wall box data via winet-s dongle not via the integrated lan port. So no two hubs accessing the inverter directly. Will try it as soon as possible.

Gnarfoz, 2024-09-24:
That should work regardless of if you talk to the WiNet-S or directly to the inverter, I think.

lassej01, 2024-09-24:
> I doubt that works with the Sungrow AC011E-01 wallbox.

No, totally with you on that one. For me a Wallbox is an own brand, I wasnt aware that Sungrow called their Wallbox, Wallbox. And the person who asked didnt state it was a Sungrow.

Louisbertelsmann, 2024-09-24:
> That should work regardless of if you talk to the WiNet-S or directly to the inverter, I think.

Unfortunately not. You can’t have two hubs on one single IP AFAIK. Homeassistant doesn’t like that.

Gnarfoz, 2024-09-24:
> No, totally with you on that one. For me a Wallbox is an own brand, I wasnt aware that Sungrow called their Wallbox, Wallbox. And the person who asked didnt state it was a Sungrow.

The link they included pointed to a german forum thread about the Sungrow product, though.
Apparently it's a language thing. In German, "Wallbox" has been referring to home charging stations for far longer than the Spanish company "Wallbox" has been around. Looks like it's one of those words that look like they're English loan words but they're really not. (Another common example: "Handy", meaning mobile phone, is not actually an English noun at all).

> Unfortunately not. You can’t have two hubs on one single IP AFAIK. Homeassistant doesn’t like that.
Why use two hubs, though?
I use a single hub (pointing to the inverter LAN port), and some sensors use slave id 1 (inverter) and some use slave id 200 (battery).
Shouldn't adding the sensors to the same hub and using the appropriate slave id (2?) work the same?

Louisbertelsmann, 2024-09-25:
> Why use two hubs, though? I use a single hub (pointing to the inverter LAN port), and some sensors use slave id 1 (inverter) and some use slave id 200 (battery). Shouldn't adding the sensors to the [...]

That’s what I already proposed earlier. I just said that it won’t work out-of-the-box when just copy pasting that into another yaml file. You will have to either copy the sensors into the existing one or use another IP to talk to the wallbox, either directly or through the inverter.

Gnarfoz, 2024-09-25:
Sorry, I must have misread that. :-)

andi-blafasl, 2024-09-28:
I got my Charger installed yesterday and followed the instructions from https://github.com/Louisbertelsmann/Sungrow-Wallbox-Modbus-HomeAssistant to setup up home assistant. For now, I see values but can't test any charging because the lag of an EV 😁

SunnyCrockett, 2024-11-06:
Maybe my forked/extended mkaiser integration is something you want to look at:
https://github.com/SunnyCrockett/Sungrow-SHx-Inverter-Modbus-Home-Assistant/tree/main/dashboards/HACS_Charger

There are all functions from the iSolarCloud app and works for me.

andi-blafasl, 2024-11-06:
I removed the modbus integration of the charger and switcht to [EVCC.io](https://evcc.io) The connection between the charger and the WiNet-S wasn't reliable and the latest update made everything worse. Don't know if this is a problem of my installation (using two RS485-to-Ethernet adapters because no cable installation possible) or with Sungrow and their Software. Before the latest update, changing anything from iSolarCloud was also very laggy and sometimes error messages were shown but changes were made anyways. With EVCC the Charging worked and the connection between EVCC and the Charger over the RS485-to-Ethernet Adapter is solid. Also I have more options how to handle and prioritize the EV and BYD Batterie. And EVCC comes with a nice addon and integration with Home Assistant. No more need to DIY your dashboard with all the values and stuff.

closed 2025-03-08 by github-actions[bot]