mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #245: MAC Adress instead IP

mega-hz, 2024-01-31:
Hi Martin,

can you manage it one can use instead of IP-Adress of ModbusTCP to use the MAC Adress?
In secrets.yaml ?

The reason i recommend this :
When Fritzbox DHCP give new IP Adresses to some Devices and Sungrow becomes a different one,
the whole Integration won't work anymore.

Another option could be to let search for AC:19:9F:**:**:** so it would find it itself?

regards,
Wolfram.

jhegewald, 2024-01-31:
Hi Wolfram,

You can configure your Fritzbox in a way, that it always assigns the same
IP to the Sungrow inverter.
(Heimnetz > Netzwerk then pick the device and you can select that option
under "Adressen im Heimnetz").

Best,
Jan

mega-hz, 2024-01-31:
Hi, 
i know this way.
at my place i made right that.
But at my neighbour that fritzbox doesn’t support this button.
 
Would it be possible to use mac?
 
Regards,
Wolfram.

mega-hz, 2024-02-02:
maybe HA can scan for the MAC-Adress and change it in secrets.yaml?

mega-hz, 2024-02-02:
BTW: wird die secrets.yaml nur einmal eingelesen oder regelmäßig?

mkaiser, 2024-02-11:
hi, 

the YAML is only parsed on HA startup (or if you select the restart option to only parse yaml).

The modbus integration does not support devices by mac address. So unfortunately, there is no way to make the MAC <> IP mapping by Home assistant. 

For finding devices in my network, I often use the freeware "Zenmap" (a GUI for nmap)

Gnarfoz, 2024-05-14:
A MAC address is not a replacement for an IP address, they work on different levels of the networking stack.
Scanning for MAC addresses, resolving them (ARP), and using the result could work, but only on the same network segment.
That would be far beyond the scope of a YAML-configuration-based integration like this and would require quite a bit of extra logic.

If your router does not offer an option to always assign the same IP address to a device (odd, haven't heard of Fritz!Box models that don't have this option), you can go the other way:
Configure your inverter to use a static IP address instead of DHCP.
I think the way to do that is to connect to the local WiFi hotspot the WiNet-S dongle provides, log into the interface there (admin / pw8888) and change the inverter's network configuration to a fixed address.
Ideally, make sure this address is _outside_ of the IP range the router uses for DHCP. You can change that range on Fritz!Boxes.

(Also: maybe that particular Fritz!Box is on an older Fritz!OS version and you first need to activate the advanced mode in the UI? I'm not sure if that particular setting is hidden behind that mode. [Maybe check the vendor documentation.](https://avm.de/service/wissensdatenbank/dok/FRITZ-Box-7362-SL/201_Netzwerkgerat-immer-die-gleiche-IP-Adresse-von-FRITZ-Box-zuweisen-lassen/))

closed 2024-11-03 by github-actions[bot]