mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #249: Some Entities are not available

kaschie2, 2024-02-03:
Describe the bug:

Some Entities are not available 

A clear and concise description of what the bug is.

Protokolldetails: (Error 1 from 5 with the same context)

TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ ( states('sensor.battery_capacity') | float * states('sensor.battery_level_nominal') | float / 100 )| round(1) }}' but no default was specified') while processing template 'Template<template=({{ ( states('sensor.battery_capacity') | float * states('sensor.battery_level_nominal') | float / 100 )| round(1) }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.battery_charge_nominal'
Your Sungrow inverter:

Model: [SH-6.RT v112]

The inverter is connected via (mark one)

[x] LAN (internal port)
[] WiNet-S (LAN)
[] WiNet-S (WLAN)
Are you using a Modbus Proxy (mark one)

[] yes
[x] no
[] I don't know what that is
Home Assistant version:
Core
2024.1.6
Supervisor
2023.12.1
Operating System
11.4
Frontend
20240104.0
modbus_sungrow.yaml:

Version/ time stamp : [e.g. 2023-12-31]
[x] I ensured to use the most recent version
** Inverter Firmware Status:**

[x] I made sure that the newest firmware is installed via the installers account
To Reproduce Steps to reproduce the behavior:

Screenshots If applicable, add screenshots to help explain your problem.
[image]

[image]
[image]

What can i do to solve the Problem? Whitout "Load Power" ist is not useable for me. 

I hope you have Ideas to help me. 

Ciao Tilo

mkaiser, 2024-02-11:
I guess the battery capacity is not set (yet). 

It should be set by the installer via isolarcloud or by the App using the "local access". It is hidden somewhere in the menues...

After setting the capacity, it should work :)

kaschie2, 2024-02-13:
In Isolarcloud the Battery ist visible and Capacity configured.

mkaiser, 2024-02-13:
Please paste following lines in "Developer tools" --> "TEMPLATE" and tell me the result:

```
{{ states('sensor.battery_capacity')}}
{{ states('sensor.battery_level_nominal')}}

{{ states('sensor.total_pv_generation')}}
{{ states('sensor.total_exported_energy')}}
{{ states('sensor.total_imported_energy')}}
{{ states('sensor.total_battery_charge')}}
{{ states('sensor.total_battery_discharge')}}
{{ states('sensor.total_consumed_energy')}}

{{ states('sensor.running_state')}}
{{ states('sensor.total_pv_generation')}}

{{ states('sensor.load_power')}}
```

It should be something like this: 
[image]

kaschie2, 2024-02-13:
[image]

mkaiser, 2024-02-13:
hmm honestly no clue at the moment ...

What happens if you comment out the modbus registers in the yaml file which are mentioned in your log file?

5008 (Inverter temperature) and 
5634 ( BMS max. charging current )

kaschie2, 2024-02-13:
Now the errors are not in Protocol but new Errors like Image 1 and Warnings like Image 2.

image 3 show one of the comment sensor. 

[image]

[image]

[image]

mkaiser, 2024-02-16:
hmm I guess there is something wrong with the communication pyModbus <> Inverter in general. 

These occurred to me, when I used multiple software connecting to the inverter at the same time (and connection request got mixed up). 

Is the HA instance at the moment the only software connecting to the inverter, or do you have something like EVCC running in parallel?

kaschie2, 2024-02-19:
Now i configured modbus Proxy and use this, but the same problem without the Problem that EVCC stoped Charging. This is now cleared with the Modbus Proxy. 

but the Sungrow Problem is not solved :(

smartmatic, 2024-04-04:
> hmm I guess there is something wrong with the communication pyModbus <> Inverter in general.
>
> These occurred to me, when I used multiple software connecting to the inverter at the same time (and [...]

I am running two HA instances Test and Prod. Ist it a problem for Modbus / Sungrow Inverter to to request data from two different instances?

Louisbertelsmann, 2024-04-05:
it is a problem, yes. For a scenario like that you might want to setup a modbus-proxy.

kaschie2, 2025-05-06:
could omebody reopen the issue? 

The Problem is not solve and maybe someone has new Ideas?

Macstar1601, 2025-05-08:
I have the same issue that some entities are not available anymore with HA 2025.5.0 installed

[image]