mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #253: Zwei Wechselrichter. Daten unterschiedlich in HA und iSolarCloud | Two inverters. Different data in HA and iSolarCloud

pille89, 2024-02-11:
**Describe the bug:**

Ich habe einen Hybridwechselrichter (SH8.0RT-V112) und einen Wechselrichter (SG5.ORT) in meinem PV System.
Leider werden die Werte scheinbar nicht zusammengefasst. Ich habe immer andere Werte in HA wie in der iSolarCloud. 
Die Werte in der iSolarCloud sind immer höher als die in HA. Kann man da etwas tun?

I have a hybrid inverter (SH8.0RT-V112) and an inverter (SG5.ORT) in my PV system.
Unfortunately, the values do not seem to be summarized. I always have different values in HA than in the iSolarCloud. 
The values in the iSolarCloud are always higher than those in HA. Is there anything that can be done?

**A clear and concise description of what the bug is.**

iSolarCloud Werte höher als in HA
iSolarCloud values higher than in HA

**Your Sungrow inverter:**
 - Model: SH8.0RT-V112 & SG5.0RT
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [x] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [x] I don't know what that is

**Home Assistant version:**
 - Version: 
Core 2024.2.1
Supervisor 2024.01.1
Operating System 11.5
Frontend 20240207.1

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2023-12-31
 - [x] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**Reproduce**
Steps to reproduce the behavior:

Besteht immer
Always exists

**Expected behavior**
Ich habe erwartet, das die Werte in HA undgefähr gleich sind wie auf dem iSolarCloud Portal

I expected the values in HA to be approximately the same as on the iSolarCloud portal

**Screenshots**
[image]

**Additional context**
Add more information about your problem here.

geni0602, 2024-02-12:
Hi,
did you try to log in locally on your inverter (Local IP) and compare with home assistant. I think the data in icloud is not live. They are refreshing in a defined period

pille89, 2024-02-12:
Hello @geni0602 ,

Thank you for your answer. How can I log on to the inverter locally? 
Do I still need a port? I can't get access via http and https.

geni0602, 2024-02-12:
i have the SG6.0rt connected with WiNetS-LAN. No Wifi active. For me it is possible to open the web interface with my local IP.

You can also go trough the app. Click on support and the LAN Local. (s. my second screenshot)

[image]

[image]

pille89, 2024-02-12:
Okay, I've been trying the Modbus IP the whole time. It doesn't work with that. It does with the WiNet-S IP. The only question that remains is whether both PV outputs are now added together. Both inverters each show a PV output that is different.

pille89, 2024-02-13:
Hello, 

So the data with two inverters is not combined. How does the script work with two inverters? When I read out the Daily PV generation, I only get the data from one inverter, the other data from the second inverter is not taken into account. 

Also die Daten mit zwei Wechselrichtern werden nicht zusammen gefasst. Wie funktioniert das Script mit zwei Wechselrichtern? Wenn ich die Daily PV generation auslese, bekomme ich nur die Daten von einem Wechselrichter angezeigt die anderen Daten vom zweiten wechselrichter werden nicht beachtet.

geni0602, 2024-02-13:
Are both invertors connected to each other? Is the first one configure to meter also the date from the second one? 

If they are independent from each other, then just configure a modbus for the second one and sum the values in home assistant 

Maybe this will help
https://www.photovoltaikforum.com/thread/184326-sungrow-zwei-wechselrichter-an-einem-smartmeter/

bleughb, 2024-02-15:
> Hello,
>
> So the data with two inverters is not combined. How does the script work with two inverters? When I read out the Daily PV generation, I only get the data from one inverter, the other data [...]

You create TWO YAML scripts, one for each inverter!
you'll have to rename all the entities in the second YAML, which can take a while (just add an _2 or similar thing to each name)
add the second IP address / password in the secrets.YAML also

I have two SG5.0RS working superbly and displaying all 4 of my MPPT's

[image]

vistree, 2024-02-24:
@bleughb - Thanx for your hints!!! Do we also need to rename each sensor - or can both inverters share the unique sensor ids?

geni0602, 2024-02-24:
> @bleughb - Thanx for your hints!!! Do we also need to rename each sensor - or can both inverters share the unique sensor ids?

Hi, it's in the name unique, one time only. They have to be different.

vistree, 2024-02-24:
Uff, that is realy hard work ;-)

geni0602, 2024-02-24:
You don't have to copy everything. Just copy the sensors that you need.

vistree, 2024-02-24:
So, do we need to rename each unique_id and each entity_id or just the entity_id?

geni0602, 2024-02-24:
[image]

geni0602, 2024-02-24:
Name and id. Entity will be created automatically according to name/id

vistree, 2024-02-24:
Thank you @geni0602 . Am I right, that I also have to rename the sensors within states?

`      - name: Battery discharging 2
        unique_id: sg_battery_discharging_2
        # TODO: test more with these parameters for template binary sensors
        # device_class: none / power
        # state_class: measurement
        availability: >-
          {{ 
          not is_state('sensor.**running_state**', 'unavailable') # e.g. here?????
          or
            (   not is_state('sensor.**ems_mode_selection_raw**', 'unavailable')
            and not is_state('sensor.**battery_forced_charge_discharge_cmd_raw**', 'unavailable')
            and not is_state('sensor.**battery_power_raw**', 'unavailable')
            and not is_state('sensor.**total_dc_power**', 'unavailable') 
            and not is_state('sensor.**load_power**', 'unavailable') 
            )
          }}
        state: >-
          {% if states('sensor.**running_state**')|is_number %}
`

geni0602, 2024-02-24:
Jep. Do you have a battery? If not, you don't need them at all.

vistree, 2024-02-24:
Yes, I have a battery ;-)

vistree, 2024-02-25:
By the way: I found https://github.com/reesaroo74/Sungrow-SHx-Inverter-Modbus-Home-Assistant which has configs for 2 inverters - which I used as a starting point, adding missing commits from mkaiser

closed 2024-10-27 by github-actions[bot]