mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #267: Battery Capacity wrong

tobaco, 2024-03-26:
**Describe the bug:**

My battery capacity shows a wrong value. It shows 0.0 instead of 9.6

**Your Sungrow inverter:**
 - Model: SH-10.RT
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [ ] WiNet-S (LAN)
   - [ ] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [ ] yes
  - [x] no 
  - [ ] I don't know what that is

**Home Assistant version:**
 - Version: 2024.3.3

**modbus_sungrow.yaml:**
 - Version/ time stamp : 2024-03-26
 - [x] I ensured to use the most recent version

**Inverter Firmware Status:**
   - [ ] I made sure that the newest firmware is installed via the installers account

**Additional context**
I tried several modbus addresses mentioned in other issues, but nothing showed the correct value (eg. 655). 

Here's my current setup (including my tried addresses):

```
      - name: Battery capacity
        unique_id: sg_battery_capacity
        device_address: !secret sungrow_modbus_slave
        address: 33048 # changed from 33048 to 13038 to 13039 to 33049 to 13038 to 33048 # reg 33049
        input_type: holding # changed from holding to input to holding
        data_type: uint16
        precision: 1
        unit_of_measurement: kWh
        device_class: energy_storage
        scale: 0.01
        scan_interval: 600
```

mkaiser, 2024-03-26:
can you check this?

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/88#issuecomment-1646586986

tobaco, 2024-03-26:
I'll try. It's something my solateur has to do. And getting him to react and to do this, will be a challenge. Thanks for the link!

mkaiser, 2024-03-26:
you can also setup a new isolarcloud account with another email adress and register yourself as "installer". Sungrow does not check this (we all did this :) )

Then you can add the plant to you as installer and have complete access to your hardware

tobaco, 2024-03-26:
Oh, I didn't knew this. Is my original installer then still in control or will he be out of the system. I don't want to jeopardize my warranty.
Thanks!

tobaco, 2024-03-26:
> can you check this?
> 
> [#88 (comment)](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/issues/88#issuecomment-1646586986)

My installer answered me in a record time (😳) and told me, he has corrected the value and also did a firmware update. 
But the sensor still shows "0,0".
I also tried the other modbus addresses, but it only changed to "655".

Louisbertelsmann, 2024-03-26:
The sensor shouldn't change, except when you expand or shrink the battery, e.g. adding or removing a module. If you aren't using the sensor, I don't see why you would want to change it. If you are using it, like for predicting the amount of charge left after a night, you can create a helper to set that capacity.

argually, 2024-04-04:
> you can also setup a new isolarcloud account with another email adress and register yourself as "installer". Sungrow does not check this (we all did this :) )
> 
> Then you can add the plant to you as installer and have complete access to your hardware

OK, New Installer Account is created, but how can one add the inverter to the new installer account? If I am trying to add it via it´s serial number that does not work.

Gnarfoz, 2024-04-04:
You need to share it from the existing customer account to the new installer account:

[image]

argually, 2024-04-05:
Thanks a lot!

closed 2024-04-05 by Louisbertelsmann