mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #268: Battery dis/charging power sensors don't work.

brixharlem, 2024-03-27:
My battery charging power and discharging power is always 0. Other battery sensors work.
This is not a new issue. I think I have it for some months already.

Your Sungrow inverter:
Model: SH-10.RT

The inverter is connected via 
 WiNet-S (WLAN)

Are you using a Modbus Proxy (mark one)
 no

Home Assistant version:

Version: 2024.3.3
modbus_sungrow.yaml:
Version/ time stamp: today

**Expected behavior**
Values show live status of battery charging and discharging power.

**Screenshots**
[image]

**Additional context**
Capacity is also 0.0. Maybe it's related?

brixharlem, 2024-03-27:
Seems to be system related. I just realised, that the values are missing in the app too.
[image]

closed 2024-03-27 by brixharlem

Louisbertelsmann, 2024-03-27:
That seems very weird. You might want to check in with you installer to make an update on battery and inverter, if you didn't install it yourself/don't want to do the update yourself.

brixharlem, 2024-03-27:
Thank you, @Louisbertelsmann. Unfortunantely my installer ran out of business and I don't have access to the firmware section.

Louisbertelsmann, 2024-03-28:
Then you can create your own installer account.

brixharlem, 2024-03-28:
The sensors are alive again! Thank you!!