mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #283: Undocumented Holding Registers destroys SH10RT Config

DemigodCode, 2024-04-12:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

The undocumented holding registers of the Inverters destroys the configuration of my inverter.
If I use one of them, they are overwritting a couple of settings and make the inverter not discharge the battery anymore until I load configuration defaults of the inverter.
The holding register discharge is not the same as the input register one.

Model: SH-10.RT v112
- [x] LAN (internal port)
- [] WiNet-S (LAN) (connected but not used for Modbus)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: 2024.1.6

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file)
 - [] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [x] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Overview'
3. Scroll down to 'Helpers'
4. Set max battery discharge power

**Expected behavior**
They should be marked as EXPERIMENTAL... That took me so much of my time to figure out what the problem is and what it did change.

**Screenshots**
[image]
[image]

mkaiser, 2024-04-19:
Hi, 

sorry to hear that!

Could you describe in more detail what happened? Strange thing is that I recently installed one SH10.RT rev112 myself for a friend. All the set* functions worked fine via GUI sliders. 

I will make some updates soon and ensure to write a comment on each undocumented sensor. 
In the current version 1.1.2 of the modbus procotoll documentation the max charging and discharging power parameters are now documented. So they should not cause your issues. 

[image]

This leaves following registers as possible culprits:
      - name: Reserved SoC for backup
      - name: Battery capacity
      - name: Battery charging start power
      - name: Battery discharging start power

Could you test max charging and discharging registers? 

Was anything regarding the other registers in the HA log files?

DemigodCode, 2024-04-19:
Can I somehow back up all settings using modbus?

I only have a user iSolarCloud account so I cannot back them up there and don't want to have my electrician to set it up again. ;-)

Don't know which parameter caused this. And don't know how to figure out.

If I change a parameter through modbus does it send the whole configuration again or the parameter to change only? I don't have much knowledge about the modbus protocol.

Do you have an idea why the holding register values differs to the other one?

DemigodCode, 2024-05-04:
Okay, I've got the time to set up a anstaller account and to create a template of my parameters.

I'm currently using ioBroker and the registers there are looking good (at least the same as in iSolarCloud)

But in HomeAssistant it's like it's not reading the parameters as it should. Logs look like that:

`
2024-05-04 22:06:02.836 WARNING (MainThread) [homeassistant.components.modbus.modbus] modbus SungrowSHx communication closed
2024-05-04 22:06:03.062 ERROR (MainThread) [homeassistant.helpers.event] Error while processing template: Template<template=({{ (states('sensor.total_imported_energy_cost')) | float |round(2) }}) renders=2>
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1984, in forgiving_float_filter
    return float(value)
           ^^^^^^^^^^^^
ValueError: could not convert string to float: 'unknown'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 567, in async_render
    render_result = _render_with_context(self.template, compiled, **kwargs)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 2305, in _render_with_context
    return template.render(**kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 1301, in render
    self.environment.handle_exception()
  File "/usr/local/lib/python3.11/site-packages/jinja2/environment.py", line 936, in handle_exception
    raise rewrite_traceback_stack(source=source)
  File "<template>", line 1, in top-level template code
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1987, in forgiving_float_filter
    raise_no_default("float", value)
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 1625, in raise_no_default
    raise ValueError(
ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.total_imported_energy_cost')) | float |round(2) }}' but no default was specified

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 693, in async_render_to_info
    render_info._result = self.async_render(
                          ^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/template.py", line 569, in async_render
    raise TemplateError(err) from err
homeassistant.exceptions.TemplateError: ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.total_imported_energy_cost')) | float |round(2) }}' but no default was specified
2024-05-04 22:06:03.074 ERROR (MainThread) [homeassistant.components.template.template_entity] TemplateError('ValueError: Template error: float got invalid input 'unknown' when rendering template '{{ (states('sensor.total_imported_energy_cost')) | float |round(2) }}' but no default was specified') while processing template 'Template<template=({{ (states('sensor.total_imported_energy_cost')) | float |round(2) }}) renders=4>' for attribute '_attr_native_value' in entity 'sensor.daily_imported_energy_cost'
2024-05-04 22:06:03.075 WARNING (MainThread) [homeassistant.components.sensor] Entity sensor.daily_exported_energy_compensation (<class 'homeassistant.components.template.sensor.SensorTemplate'>) is using native unit of measurement '€' which is not a valid unit for the device class ('energy') it is using; expected one of ['MJ', 'MWh', 'Wh', 'kWh', 'GJ']; Please update your configuration if your entity is manually configured, otherwise create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+template%22
`

And now to show you what the difference is:

[image]

[image]

Am I missing something?

mkaiser, 2024-05-06:
Cannot find **(states('sensor.total_imported_energy_cost'))** in this repositories YAML file. I guess the error comes from somewhere else. Can you double-check?

Gnarfoz, 2024-05-06:
> ![image](https://private-user-images.githubusercontent.com/36764562/327973632-178b9d62-d50f-4d8b-862a-c81bb5f70778.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ5ODU2MDksIm5iZiI6MTcxNDk4NTMwOSwicGF0aCI6Ii8zNjc2NDU2Mi8zMjc5NzM2MzItMTc4YjlkNjItZDUwZi00ZDhiLTg2MmEtYzgxYmI1ZjcwNzc4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA2VDA4NDgyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg2YmQ0MjJhMTVlZjBlZGExYjdmNzRjZTBlYzRjNmEwMmYwZGM2ZTQzZmU0M2FlZmZjYzNhYmU5ZjczYmViMTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.m0O-pNCt8_PIHzlL7ge7ZCvzPX0v1IlGUT6FoxczklQ)
> 
> ![image](https://private-user-images.githubusercontent.com/36764562/327973642-684a53d0-0352-4853-8081-55d0a35d16b1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ5ODU2MDksIm5iZiI6MTcxNDk4NTMwOSwicGF0aCI6Ii8zNjc2NDU2Mi8zMjc5NzM2NDItNjg0YTUzZDAtMDM1Mi00ODUzLTgwODEtNTVkMGEzNWQxNmIxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTA2VDA4NDgyOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI1ZTQwYzIyYmRlZWEyNjY3MjBhOWUwYWIzMDUxYTY2ZWYwMjIxYzBmZGY3YmJmNWFmNzhmOGM2MmJjYzI1YmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.ij_HCwdC5Xh_VdU3fUrzJhJZoL96J52_cigmNmFB9cI)

The max battery charge / discharge entities come with a default maximum of 5000 W since @mkaiser can't know your actual battery size / parameters.
You should adjust it to match your actual battery parameters. I assume it will then display the correct value.

You can find them here: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2325
and here: https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml#L2331

jbardu, 2024-07-30:
regarding installers accounts to enable advanced settings and backup @mkaiser  -- I can create an installers account and login on it -- but how do I attach my existing plant to it, in order to do anything?

When I "create" a plant and then enter a serial number of the comms dongle it says (rightly) it is already taken. thanks.

Gnarfoz, 2024-07-30:
You share the plant from your regular account to your installer account.

closed 2024-12-08 by github-actions[bot]