mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #291: Modbus no longer works

Hirates, 2024-05-06:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**

A clear and concise description of what the bug is.

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT v112]
 - The inverter is connected via (mark one)
   - [192.168.1.240] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [] no 
  - [X] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2024.05.1]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file)
 - [X] I ensured to use the most recent version

** Inverter Firmware Status:**
   - [X] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
I have been looking at developer options for a sensor and there is not available (Every Modbus sensor) 

LOG:

Logger: homeassistant.components.modbus.modbus
Quelle: components/modbus/modbus.py:331
Integration: Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 5. Mai 2024 um 23:34:36 (1 Vorkommnisse)
Zuletzt protokolliert: 5. Mai 2024 um 23:34:36

Pymodbus: SungrowSHx: Error: device: 1 address: 4999 -> Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.1.240:502]

**Expected behavior**
Can anyone tell me what is going on? I have checked the IP address of the inverter and restarted both Hass and the inverter, but I still get the error message.

I have never had similar errors, I have not changed a single bit of the modbus, but yesterday it stopped working

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

mkaiser, 2024-05-06:
Can you check with QModBusMaster if the connection is working in general?

https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/wiki/FAQ:-Problems-with-the-connection#check-3

Hirates, 2024-05-06:
[image]

Thanks for the prompt reply, here is a screenshot.
I think it isn't sending any data.

I've tested it with 192.168.1.240 and 192.168.001.240 both the same result.

mkaiser, 2024-05-06:
Looks like the IP address of the inverter has changed. 

Can you access your router and check? If you have the IP use a terminal / cmd prompt and ping the inverter:

should look like this: 

[image]

Gnarfoz, 2024-05-06:
If the address is unchanged, do you maybe have more than one thing accessing the inverter via Modbus?
Even if not, the inverter's Modbus interface is finnicky and prone to just giving up entirely if it encounters anything it doesn't like (mostly: overlapping Modbus queries).
A trivial remedy appears to be unplugging the network cable for a moment.

Hirates, 2024-05-06:
I have changed the subnet of the inverter in my UDM Pro for further testing. 
This caused the inverter to reconnect to the router. This solved the problem.

Thanks for your fast reply :)

closed 2024-05-06 by Hirates

tkahn, 2024-05-08:
I had a similar error and it was also related to a **subnet setting**. 

In Home Assistant (running on a Raspberry Pi) the IP was set to **192.168.1.XXX/32** which effectively made Home Assistant think it was all alone on its own subnet. It will assume that there are no other hosts within its subnet with which it can communicate directly (effectively a subnet setting of 255.255.255.255). The actual subnet setting of my network is 255.255.255.0. By changing this setting in Home Assistant to **192.168.1.XXX/24**, Home Assistant got the correct subnet setting (255.255.255.0) and found the Sungrow inverter immediately after having rebooted Home Assistant.

mkaiser, 2024-05-08:
@tkahn: good to know, thank you for the feedback!