mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #292: Some values ​​are missing after updating to 2024.5.2

MeinerB, 2024-05-07:
Hello,
I waited to update the HA until the problems with the 2024.4.x version were fixed.
Yesterday I updated everything, unfortunately I'm missing some values.

All symbols in the inverter state are incorrect because the "sensor.running_state" is obviously set to 0.

The following values ​​are set to 0:
- sensor.battery_discharging_power
- sensor.battery_charging_power

- sensor.grid_frequency show 500Hz 

The following values ​​are not available:
sensor.total_backup_power
sensor.backup_phase_a_power
sensor.backup_phase_b_power
sensor.backup_phase_c_power

In my opinion everything is up to date except for the inverter firmware, which is still the one from the installation from June 23rd

Thank you for further help,
Bernd

**Your Sungrow inverter:**
 - Model: SH10RT-V112
 - The inverter is connected via (mark one)
   - [x] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [] yes
  - [x] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: [2024.05.2]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [2024-04-05] (see header of the file)
 - [x] I ensured to use the most recent version

MeinerB, 2024-05-10:
Es scheint so als ob er den "sensor.running_state" nicht auslesen kann und deswegen die Werte nicht füllt.

Unter Templates wird folgendes angezeigt:
{{ states('sensor.running_state')}} = 0   (Seit dem Update und dem Beginn der Probleme hat sich der Wert nicht geändert)
{{ states('binary_sensor.battery_charging')}} zeigt OFF
{{ states('binary_sensor.battery_discharging')}} zeigt OFF
{{ states('sensor.battery_power_raw')}} zeigt einen numerischen Wert (W) mit dem offensichtlich die Batterie geladen wird.

chstoltz, 2024-05-21:
Diesen Fehler hatte ich auch, bei mir hat ein aus- und einstecken des Netzwerkkabels am Hybrid WR geholfen.

Dennis1987m, 2024-06-22:
> Diesen Fehler hatte ich auch, bei mir hat ein aus- und einstecken des Netzwerkkabels am Hybrid WR geholfen.

Funktioniert bei mir leider nicht.

dylan09, 2024-06-23:
Gibt es denn Meldungen der ModBus Integration im HA Log?

Dennis1987m, 2024-06-24:
Ich kann damit zwar nichts anfangen, aber das hier habe im im Log gefunden:

Logger: homeassistant.components.modbus.modbus
Quelle: components/modbus/modbus.py:331
Integration: Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 22. Juni 2024 um 11:13:32 (17089 Vorkommnisse)
Zuletzt protokolliert: 09:42:49

Pymodbus: SungrowSHx: Error: device: 1 address: 6226 -> pymodbus returned isError True
Pymodbus: SungrowSHx: Error: device: 1 address: 6230 -> pymodbus returned isError True
Pymodbus: SungrowSHx: Error: device: 1 address: 6275 -> pymodbus returned isError True
Pymodbus: SungrowSHx: Error: device: 1 address: 5601 -> pymodbus returned isError True
Pymodbus: SungrowSHx: Error: device: 1 address: 5725 -> pymodbus returned isError True

dylan09, 2024-06-24:
Die ersten 3 ist die monatliche/jährliche Erzeugung. Die komplett mal auskommentieren (Jan. bis Dez.) und die jährlichen.
Wie oft werden die gepollt?

Dann mal prüfen, ob es besser wird, wenn bei den Registern, die im 10 Sekunden Takt gepollt werden der Fehler weg ist, wenn du das Intervall verlängerst?

Aber keines der Register aus dem Log haben etwas mit dem Status des Inverters und Charging Power zu tun.

Remigius0815, 2024-09-01:
Hänge mich hier mal mit dran, gleiches Problem....

DarcDan, 2024-09-01:
Hier auch ;-)

closed 2025-01-05 by github-actions[bot]