mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant #295: sungrow update no battery forced mode

Inrebusillis, 2024-05-10:
**Before you create an issue, make sure to update to the current version of [modbus_sungrow.yaml](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/modbus_sungrow.yaml)**

**Describe the bug:**
Setting the battery forcedcharge  with the isolarcloud is ok
no change is setted with the GUI.

A clear and concise description of what the bug is.
I am trying to force the battery to charge via the GUI or via register 13050 direct MODBUS command.
I noted with MODBUS reader  that the register13050 value stay on 00XAA in both case  I put to forced or default mode the system via isolarcloud ( in this case the battery is forced to load and doing it)

**Your Sungrow inverter:**
 - Model: [e.g. SH-10.RT v112]
 - The inverter is connected via (mark one)
   - [YES] LAN (internal port)
   - [] WiNet-S (LAN)
   - [] WiNet-S (WLAN)

- Are you using a Modbus Proxy (mark one)
  - [YES] yes
  - [] no 
  - [] I don't know what that is

**Home Assistant version:**
 - Version: [e.g. 2024.05.2]

**modbus_sungrow.yaml:**
 - Version/ time stamp : [e.g. 2023-09-08] (see header of the file)
 last update: 2023-07-08
 
 No, latest version return to me device _ address not valid...

 - [] I ensured to use the most recent version

** Inverter Firmware Status:**
[image]

   - [YES] I made sure that the newest firmware is installed via the installers account

**To Reproduce**
Steps to reproduce the behavior:
1. Go to isolarcloud
2. Click on energymanagement change to force charging
3. open MODBUS reader
4. See error

**Expected behavior**
register 13050 shold change from 00XAA to 00XBB or 00 X CC
it stay 00XAA

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

Gnarfoz, 2024-05-11:
For me the following works:
While the "EMS mode selection" (register 13050) is on "self consumption mode" (0), the "charge/discharge command" (register 13051) toggle is set to "stop" (0xCC).
If I want to force charge/discharge, I select "forced mode" (2) for "EMS mode selection" and "charge" (0xAA) or "discharge" (0xBB) for the "charge/discharge command".
They're the two dropdowns provided on the [default dashboard offered in this repository ](https://github.com/mkaiser/Sungrow-SHx-Inverter-Modbus-Home-Assistant/blob/main/dashboards/_DefaultDashboard_mkaiser/3_EMS_Control.png).

Regarding register 33208, considering it directly precedes 11 other registers (33209-33219) related to scheduled forced charging, I'm inclined to assume it toggles the _time based_ forced charging and is not related to the above "forced mode" method.

Inrebusillis, 2024-05-12:
Thanks! that make perfectly sense, I was not aware that I shall change first EMS mode and then the batterie forced as you describe.
Much simple to integrate with home automation and Tibber as you just need to set this 2 parameter.

closed 2024-05-12 by Inrebusillis